RethinkRobotics-opensource/rosnodejs #4: Proper acknowledgement

odestcj, 2016-04-21:
Please respect the original rosjs creators with acknowledgement citations (e.g., [Osentoski et al. 2011 ICRA]):

http://ieeexplore.ieee.org/xpl/login.jsp?tp=&arnumber=5980464&url=http%3A%2F%2Fieeexplore.ieee.org%2Fxpls%2Fabs_all.jsp%3Farnumber%3D5980464

Thank you,
-Chad

IanTheEngineer, 2016-04-21:
Hi Chad.
Does `rosjs` still exist? I thought it was now `roslibjs` was preferred. You have it marked as deprecated here: http://wiki.ros.org/rosjs
This package the name `rosjs`, with no code (or implementation) shared between. It is quite different in that it is a native ROS client implementation, without requiring a bridge to access the ROS network.

That said, we very much respect your work with `RobotWebTools`, and we intend to make it very clear in the documentation that this is a separate and distinct package. Please let us know your thoughts.

Best,
~ Ian

chris-smith, 2016-04-21:
We have considered renaming this `rosnodejs` instead since it is, currently, targeted to run in node. There's an existing deprecated [`rosnodejs`](https://github.com/baalexander/rosnodejs) package as well though. Renaming it to `rosnodejs` would also be a little strange if we ever get to the point of trying to run this outside of node. Admittedly we may not, but I think that was what had originally swayed us.

odestcj, 2016-04-21:
Thanks, Ian.  I really appreciate your message and quick reply.  I should also said and emphasized that I have appreciated your and Rethink's use of and contributions to RWT, which has been great to see.  The use of Figure 1 of this paper was meant to give tribute to Rethink:

http://ohseejay.org/papers/rctoris_iros2015.pdf

The issue is partially that we have a naming conflict with our past publications, and a need for citations that reflect the impact of our work.  Many people have contributed much time, thought, and effort to build up rosbridge, the various ROS JavaScript libraries, RWT, and other open-source robotics releases without much recognition.  Gaining status and fanfare has never been the goal, but a strong desire to help advance robotics and human-robot interaction in a new direction.  As needs develop and change, I am thrilled to see new code and ideas appear that emphasize the glaring deficiencies of communications for robotics and cloud robotics.  But, a brother still needs some props. 

That said, we still need some form ack to justify our efforts that show we have had an impact to our colleagues, universities, funders, etc.  Given the respect and reach of Rethink, such acks in a repo for this project go a long way to helping contributors to open-source robotics feel good about their efforts, want to contribute more in the future, and find sustainable pathways to support such projects (at least for those of us who do not have OSRF-level bankrolls). Thus, it would be great if the repository repo could cite the relevant papers and repos that have contributed/inspired this project.  For example, I appreciated that the ardrone-autonomy ROS package still gives an ack to our older ardone-brown package, which SFU built upon:

  http://ardrone-autonomy.readthedocs.org/en/latest/

So, maybe there could just be an acknowledgements section in the README.md with pointers to relevant papers and repos, similar to lists of contributors.  And to be fair, I realize there is little in the way of precedent or common practices for conducting such citations for open source.  I hope to see and develop such practices that become commot to both the robotics research and open-source community, for which this issue is a start.

Thanks again!
-Chad

odestcj, 2016-04-21:
I forgot to address your question.  The original rosjs does not have an active repo beyond _rosbridge_ v1 and ROS Electric.  However, rosjs does still exists in published papers.  

I do not really mind if you continue using "rosjs" with attribution.  My core issue is somewhat orthogonal.  I am really getting at the culture of attribution for contributions from research involving open-source robotics.  In this regard, my issue is with the ROS developers and maintainers who have not created a culture of proper acknowledgement (partially through its overemphasis of the BSD license), which has created an imbalance in attribution.  I just believe that the creation and maintenance of healthy open-source ecosystem relies on proper acknowledgement.  As such, I can only ask third-party ROS repositories to ack ideas and code that served as inspiration and/or sources of code.

chris-smith, 2017-11-27:
README was updated in August in [this commit](https://github.com/RethinkRobotics-opensource/rosnodejs/commit/893774b2bf8a2d68241c51d0b44605ddc93bad83).

closed 2017-11-27 by chris-smith