RethinkRobotics-opensource/rosnodejs #6: Create message classes on the fly

chfritz, 2016-04-25:
Hey guys,

rosnodejs had this nice ability to find and load .msg files for the used message types and generate the appropriate js class definitions on the fly -- and I extended that to .srv files, too. This essentially brings what you currently seem to have in genjs into the library itself, i.e., it saves a compilation step. I was thinking of porting that functionality to rosjs. Are you open to pull requests like that?

In general I'd like to contribute to rosjs if I can. I was just pointed to this project by Brandon Alexander, who wrote rosnodejs. I was going to complete that project, but it seems that you guys are way ahead of me and already rewrote the whole thing in super clean ES6 :-)

Anyhow, please let me know how you feel about contributions in general and this particular feature in particular.

chris-smith, 2016-04-25:
Hey Christian!

> contributions in general

Contributions are great! As far as it currently goes, most things seem to _work_ but we're still missing functionality typical of ros nodes. I'd also just love to get more hands on it because presumably there are bugs I haven't found yet.

> this particular feature in particular.

I think this would be pretty great. As far as I'm aware, this would require users to pull down any default message packages they want to use into their repositories, right? I'd prefer not to make that a requirement for people using rosjs, but I do think it would be a great use case to support. It would be really nice to provide support for people working outside a ROS environment or whose development/build/etc interests are different from our own - as of right now we're just supporting a more typical catkin-ROS environment.

chris-smith, 2016-04-26:
Actually, the raw message files apparently do exist. Depending on the implementation and people's use cases it's possible they would prefer the compiled version, but I can't see a reason not to support both. Presumably the hooks would go in message_utils and users would set some flag to specify which version they're using?

chfritz, 2016-04-28:
Yeah I think that's a good approach. Presumably the pre-compiled and the on-the-fly compiled will be identical anyways, but allowing caching on disk of already compiled classes makes sense too. They all have md5 sums anyways to verify that a user is using the right one, so identity is easily established. And yes, rosnodejs finds the raw definition files (.msg and now also .srv) in the ros file tree, parses it, and generates the class definition on the fly. So that's what I was planning on porting.

PR #12 (2016-05-18): Issue#6 generate message handlers automatically
Fixes #6.

This PR ports the message handler generation from @baalexander's rosnodejs, extended for services in https://github.com/chfritz/rosnodejs, to this repo. The newly added example2.js shows how to run this.

As a reminder, the main benefit of this is that it doesn't require gennodejs to have already generated javascript class definitions. Instead these classes are generated on demand (and in javascript itself). In the current implementation such demand is expressed using the `rosnodejs.use(messageTypes, serviceTypes, callback)` function ( [... cut]
closed 2016-05-18 by chris-smith