RethinkRobotics-opensource/rosnodejs #7: Rename rosjs

chris-smith, 2016-04-25:
We're thinking about renaming `rosjs` but are unsure of what to call it. The name `rosjs` originally belonged to a [package](http://ieeexplore.ieee.org/xpl/login.jsp?tp=&arnumber=5980464&url=http%3A%2F%2Fieeexplore.ieee.org%2Fxpls%2Fabs_all.jsp%3Farnumber%3D5980464) developed at Brown for Rosbridge 1.0 - our repo is also specifically targeting Node.js so the name `rosjs` could be misleading. `rosnodejs` seems like an obvious answer, but there is also an existing deprecated [rosnodejs](https://github.com/baalexander/rosnodejs) project.

Our attempts to follow ROS client library naming conventions have apparently led us into a trap.

Have any thoughts, @baalexander?

baalexander, 2016-04-26:
I like what you all are doing with this package and have been impressed with your code quality. What is your plans going forward with rosjs (or whatever name it takes on)?

I ask because I know there's been interest in an official ROS node.js package by others. For example @chfritz asked about the same thing on [this issue](https://github.com/baalexander/rosnodejs/issues/37#issuecomment-213163249) a few days ago. I personally think the Rethink package is off to a great start and I would be happy to transfer the rosnodejs name to you all, I just want to check what your plans are going forward.

chris-smith, 2016-04-27:
At Rethink, we're planning to transition to the native implementation in the near future. At that point, maintenance of the code will be very important to us. From our perspective, opening it up to the community for use, contribution, and maintenance can only help. We'd also prefer not to fracture the community across multiple implementations or to have the community migrate to a separate package and be left with this on our own.

@IanTheEngineer and I are currently the only ones working on this. This is mostly work that I've done in my spare time so far too - as we move towards migrating over I expect that will change more. Once its in our code base I expect other Rethink-ers will actively contribute to features, bugs, and maintenance as well - @IanTheEngineer, @rethink-rlinsalata, and a few others have dedicated a fair amount of time at work to investigating and solving [bugs](https://github.com/ros/ros_comm/issues/577) in other core ROS libraries too. My goal right now is to make sure the transition will be smooth and we don't inadvertently break things as new functionality goes in - so I'm mostly writing tests.

baalexander, 2016-04-27:
I'm happy to see you all are open to community contributions.

I am okay with transferring the rosnodejs name and NPM package to you and @IanTheEngineer.

PR #10 (2016-05-09): Rename rosjs to rosnodejs
Since the message generator was renamed to `gennodejs` and @baalexander graciously offered the name `rosnodejs` for this ROS Javascript library, this pull request changes all of the references of `rosjs` to `rosnodejs`. I will rename the repo once we merge. Resolves #7
closed 2016-05-09 by chris-smith