RethinkRobotics-opensource/rosnodejs #9: Complete Testing Framework

IanTheEngineer, 2016-04-29:
- [ ] Add ROS Actions to test scripts
- [ ] Create a `test_srvs` package of complex ROS Services
- [ ] Add ROS Actions to test scripts
- [ ] Build in automated Jenkins testing hooks to invoke these tests on a new pull request to either `rosjs` or `gennodejs`

NickZ, 2017-12-08:
I notice you guys have set up Travis CI; I think this would be fairly easy to implement on there?

chris-smith, 2017-12-12:
Should be - setting up the environment and actually running the tests there has been on my list for a while...

louisgv, 2018-07-09:
@chris-smith  What else does this issue need to close?

chris-smith, 2018-07-10:
We do automatically run tests against pull requests/commits through travis...

It'd be good if we tested against future versions of Ubuntu - I think we need docker for this because travis only supports 16.04. @IanTheEngineer was planning to look into this recently but I'm not sure if he got an opportunity.

louisgv, 2018-07-13:
I have been investigating about init Docker via Travis and reading their documentation. After further planning, I will submit a PR that have Travis spin up docker for several distro and run the test.

@chris-smith Can you give me a list of target distros we would like to test against?

chris-smith, 2018-07-23:
I think 14.04, 16.04, and 18.04 would be good - support for 14.04 will probably drop ~soon (I think OSRF is dropping 14.04/Indigo next April) but we should continue supporting for now.

louisgv, 2018-07-27:
I went with Travis's build matrix as it seems that it is supporting later Ubuntu versions. However, it seems that xenial and bionic cannot locate ros indigo... And for some reason trusty is failing on some dpkg call :\ ...

chris-smith, 2018-07-27:
I think we'll need to use different ROS versions for xenial and bionic too (probably kinetic and melodic respectively)

louisgv, 2018-07-30:
@IanTheEngineer  I would love to dive into dockerizing the CI and try my best to help. Would you happen to have a list of material / convention / dockerizing method that you would like me to go through and follow for this issue?

I was looking at maybe we can use docker compose to ease the scripting process :-?

louisgv, 2018-08-13:
[image]

I managed to set custom ROS version per distro, and per the CI log above, it succeed for trusty and xenial, but failed on bionic. The error message on `bionic` is:

```
E: Unable to locate package ros-melodic-catkin
```

Seems like the pipeline to work with catkin changed in melodic?

It is also seems travis does not play well with multidimensional matrix, thus I went ahead and manually specify all the nodejs version and Ubuntu distro combination. The latest travis build should reflect that.

LMK if you have any feedback! @IanTheEngineer @chris-smith

chris-smith, 2018-08-20:
I'm not immediately sure why `ros-melodic-catkin` can't be found... I'm a little surprised the 16.04 and 18.04 builds start trying to install packages though. As far as I'm aware, travis doesn't actually support either of them. @IanTheEngineer seems to have made different branches for each release in [sns_ik](https://github.com/RethinkRobotics-opensource/sns_ik/blob/indigo-devel/.travis_ci/travis.sh), along with docker images for each. I'd prefer not to have to maintain different branches for each ROS release if possible though.

louisgv, 2018-08-20:
I inspected the log again and apparently Travis does not have 18.04 for their matrix yet, and thus it fallback to 12.04: 
[image]

I will setup matrix for env + docker using https://hub.docker.com/r/rfc1149/rose-dev/ and we will see if that works.