RethinkRobotics-opensource/rosnodejs #14: Unify API for on-the-fly generated and pre-compiled message classed

chfritz, 2016-05-18:
See https://github.com/RethinkRobotics-opensource/rosnodejs/pull/12#issuecomment-219899631.

This is complicated slightly by the fact that pre-compiled message classes can be instantiated synchronously, whereas on-the-fly generation of message classes is asynchronous. So a unified interface may need to follow the asynchronous approach, i.e., use a callback rather than a return value.

chris-smith, 2016-05-19:
If you promisify on-the-fly message generation (or at least part of it), you could do the asynchronous generation in `rosnodejs.init` e.g.
```
rosnodejs.init('/my_node', {messages: ['std_msgs/String'], services: ['std_srvs/SetBool']})
.then(...)
```

I think the messages should probably be generated before we start checking for the master. So where we create a new promise in the init function, if users provided messages and services in the options, you would instead do
```
init() {
  ...
  return this.use(options.messages, options.services)
  .then(() => {
    _checkMasterHelper(resolve, 0);
  }
}
```

chfritz, 2016-05-19:
That sounds good!

chfritz, 2016-05-19:
OK, PR created.

PR #32 (2016-09-27): Unify interface further
Further unified the interfaces of on-the-fly and gennodejs.

See the updated example_turtle.js for the updated syntax. Basically, the specification of which messages and services to use is no longer necessary in the on-the-fly approach. We just generate *all* message and service definitions on start, which only takes about 500ms. After that they can be accessed just like the gennodejs generated ones using `rosnodejs.require`.

Fixes #14.

Also implemented uint64 and int64 support.
- tested with rostopic; it works, with the usual caveat that
  javascript only uses 53bit  [... cut]
commit e6cd0376 (2016-09-27): Further unifying the interfaces

Implemented uint64 and int64 support

Now generating all messages and services on the fly on start (if requested)

- message.getAll works and is very fast (takes less than a second)

- Fixes #14.

Corrected implemention for [U]Int64 in on-the-fly approach

Using bignum for [u]int64 if available, otherwise usual javascript
numbers are used but a warning is printed if numbers are sent or
received that are outside of the precise range of JS for integers (>=
1e15).
closed 2016-09-27 by chris-smith