RethinkRobotics-opensource/rosnodejs #28: npm test fails; DeserializeStream components give "not implemented"

iikkateivas, 2016-07-05:
Hi,

I'm interested in creating a ROS web client and this seems to be the only way to do it while using Kinetic. However, there is something wrong with this since the npm test fails on 1) Logging Rosout Check Publishing and so on.  Here is a part of the npm test output: 

  9 passing (17s)
  18 failing

  1) Logging Rosout Check Publishing:
     Uncaught Error: not implemented
      at DeserializeStream.Transform._transform (_stream_transform.js:142:9)
      at DeserializeStream.Transform._read (_stream_transform.js:167:10)
      at DeserializeStream.Transform._write (_stream_transform.js:155:12)
      at doWrite (_stream_writable.js:292:12)
      at writeOrBuffer (_stream_writable.js:278:5)
      at DeserializeStream.Writable.write (_stream_writable.js:207:11)
      at Socket.ondata (_stream_readable.js:528:20)
      at readableAddChunk (_stream_readable.js:146:16)
      at Socket.Readable.push (_stream_readable.js:110:10)
      at TCP.onread (net.js:523:20)

  2) DeserializeStream basic:
     Error: not implemented
      at DeserializeStream.Transform._transform (_stream_transform.js:142:9)
      at DeserializeStream.Transform._read (_stream_transform.js:167:10)
      at DeserializeStream.Transform._write (_stream_transform.js:155:12)
      at doWrite (_stream_writable.js:292:12)
      at writeOrBuffer (_stream_writable.js:278:5)
      at DeserializeStream.Writable.write (_stream_writable.js:207:11)
      at Context.<anonymous> (test/DeserializeStream.js:22:23)

  3) DeserializeStream 2 chunks:
     Error: timeout of 2000ms exceeded. Ensure the done() callback is being called in this test.

Any ideas? Is there something I'm missing here?
- Iikka

chris-smith, 2016-07-05:
Hmm... well if the `DeserializeStream`'s not working almost nothing is going to work. What version of `node` are you using?

The other option for ROS web clients is [rosbridge](http://wiki.ros.org/rosbridge_suite). I'd love for people to use `rosnodejs` but `rosbridge` will be more stable at this point.

iikkateivas, 2016-07-05:
I'm using node 4.2.6 which should be the LTS version. 

Is the rosbridge working on Kinetic yet? I was also reading about that but I found only support for indigo.

chris-smith, 2016-07-05:
That's probably it. I've been working against 5.2 and have been meaning to transpile through `babel` at some point - the issue just hadn't come up yet.

iikkateivas, 2016-07-06:
Thank you for your quick answers. You were right about that node version. 5.2 works fine.

chfritz, 2016-09-15:
This was fixed by https://github.com/RethinkRobotics-opensource/rosnodejs/pull/31.

IanTheEngineer, 2016-09-15:
Closing thanks to @chfritz contribution

closed 2016-09-15 by IanTheEngineer