RethinkRobotics-opensource/rosnodejs #42: "rosnode info" on a rosnodejs node fails to contact

IanTheEngineer, 2016-12-09:
This is the console output from my ROS workspace:
```
$ rosnode info /my_rosnodejs_node
--------------------------------------------------------------------------------
Node [/my_rosnodejs_node]
Publications: 
 * /chatter [std_msgs/String]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: None

Services: 
 * /my_rosnodejs_node/get_loggers
 * /my_rosnodejs_node/set_logger_level

contacting node http://Artoo.local:39868 ...
ERROR: Communication with node[http://Artoo.local:39868] failed!
```
and the console.log reads the following:
```
ConsoleLogStream.js:56 [ERROR] [1481253678.610] (ros.rosnodejs): Not implementedwrite @ ConsoleLogStream.js:56Logger._emit @ bunyan.js:938(anonymous function) @ bunyan.js:1037_handleGetBusInfo @ RosNode.js:524emitThree @ events.js:116emit @ events.js:194(anonymous function) @ server.js:42callback @ deserializer.js:65Deserializer.onDone @ deserializer.js:92emitNone @ events.js:86emit @ events.js:185SAXStream._parser.onend @ sax.js:190emit @ sax.js:640end @ sax.js:683end @ sax.js:154SAXStream.end @ sax.js:248onend @ _stream_readable.js:511g @ events.js:286emitNone @ events.js:86emit @ events.js:185endReadableNT @ _stream_readable.js:974_combinedTickCallback @ internal/process/next_tick.js:74_tickCallback @ internal/process/next_tick.js:98
ConsoleLogStream.js:50 [DEBUG] [1481253680.359] (ros.rosnodejs): Publishing message on /chatter with no subscribers
```

cinderblock, 2017-01-06:
Which version of `rosnodejs` are you trying this with? I was having problems getting the version that is published on npm to work. See #44

chris-smith, 2017-01-13:
This is almost certainly because I didn't get to implementing `getBusInfo` or `getBusStats`...

Relevant code is at `lib/RosNode.js l521`

[`ROS Slave Api`](http://wiki.ros.org/ROS/Slave_API)

chris-smith, 2017-02-26:
This is now resolved by #51

closed 2017-02-26 by chris-smith