RethinkRobotics-opensource/rosnodejs #44: Version published on npm seems to be broken - where to get stable version?

cinderblock, 2017-01-06:
I was trying to get some simple examples working based on the currently published npm version (claims to be `1.0.0`).

For the life of me, I could not get even the simplest messages to send or be received without node crashing.

#### Publisher
```
cameron@AuntPolly:~$ rostopic pub /topic std_msgs/String test
```
#### Subscriber
```
cameron@AuntPolly:~/ros/node-test-recv$ npm start

> node-test@0.0.0 start /home/cameron/ros/node-test-recv
> babel-node .

[/test-rcv][INFO] 2017-01-04 23:40:39.426-0800: Connecting to ROS Master at http://localhost:11311
[INFO] 2017-01-04 23:40:39.962-0800: Connected to master!
[sub/topic][WARN] 2017-01-04 23:40:39.980-0800: Error requesting topic: Error: connect ECONNREFUSED 127.0.1.1:60412, undefined
test
[sub/topic][INFO] 2017-01-04 23:40:48.703-0800: Pub AuntPolly:37577 sent END
[sub/topic][INFO] 2017-01-04 23:40:48.704-0800: Pub AuntPolly:37577 closed
/home/cameron/ros/node-test/node_modules/xmlrpc/lib/server.js:41
      if (that._events.hasOwnProperty(methodName)) {
                       ^

TypeError: that._events.hasOwnProperty is not a function
    at /home/cameron/ros/node-test/node_modules/xmlrpc/lib/server.js:41:24
    at Deserializer.callback (/home/cameron/ros/node-test/node_modules/xmlrpc/lib/deserializer.js:65:7)
    at Deserializer.onDone (/home/cameron/ros/node-test/node_modules/xmlrpc/lib/deserializer.js:92:12)
    at emitNone (events.js:86:13)
    at SAXStream.emit (events.js:185:7)
    at SAXParser.SAXStream._parser.onend (/home/cameron/ros/node-test/node_modules/sax/lib/sax.js:184:8)
    at emit (/home/cameron/ros/node-test/node_modules/sax/lib/sax.js:615:33)
    at end (/home/cameron/ros/node-test/node_modules/sax/lib/sax.js:654:3)
    at SAXParser.end (/home/cameron/ros/node-test/node_modules/sax/lib/sax.js:149:24)
    at SAXStream.end (/home/cameron/ros/node-test/node_modules/sax/lib/sax.js:234:16)
```

The exact sequence / set of messages printed changes based on if published or receiver is started first, if roscore is restarted before, or if I use a `rosnodejs` based trivial publisher. I can elaborate on this if requested.

I then tried using the `kinetic-dev` branch here. That gave very different errors. Probably because it's not a stable branch. Moving on...

I finally tried the version tagged `1.1` in git (which according to `packages.json` is version `1.0.0` still) and the simple finally examples worked as expected.

1. Could the most recent stable version of this be published to npm?
2. What is the current recommended place/way to get stable code?

chris-smith, 2017-01-06:
Unfortunately what "works" and how it works is for now largely a function of what version of node and ROS you're using. The version tagged `1.1` in git was mostly just tagged as a way to mark the history since it was directly before a large merge that included lots of changes was pulled in and less a version that we planned to release  - I would expect most things to work.

The version released to `npm` will work with generated messages from ROS kinetic out of the box right now, while none of the current versions will. We've been meaning to release a new version of `gennodejs` into kinetic and then push a new version of `rosnodejs` to npm but haven't had time to do all the testing we need to before releasing `gennodejs`.

The `kinetic-devel` branch should be relatively safe though. None of the current issues that we're aware of cause any ROS pub/sub, service instabilities. I think it's been rare that we've pulled in things that have broken previously working functionality. I have no problems running `kinetic-devel` in node 6.2.2 or 6.9.1 right now. Testing against other versions of node has also been in the backlog for a while so ymmv... I remember seeing that events issue before and I think it was due to node versions - though I can't currently recall which versions worked/didn't.

chris-smith, 2017-01-06:
As another side note, possibly the most stable versions are actually from a fork of mine (and its associated releases) at [indigo-devel](https://github.com/chris-smith/rosjs/tree/indigo-devel) - this is the branch Rethink has been using for a few months now. The large merge into `kinetic-devel` that I mentioned was my merge of the `indigo-devel` branch that I created to separately handle what I expected to be the mess of issues that arose in attempting to actually deploy `rosnodejs` in a large development environment (as opposed to just small examples I had done previously) and in trying to adapt it to work in `Indigo` where messages aren't created as a part of the normal ROS build like they are in `Kinetic`.

We're hoping to migrate to ROS `Kinetic` soon and planning to move off my fork then. The `Indigo v1.0.5` release has, I believe, thousands of hours of run time without issues against node 6.2.2.

cinderblock, 2017-01-06:
I've added my versions to the original post for reference.

I'd like to use all the newest stuff. I don't want to get stuck using some old system. I'd also like to stay on Kinetic as it's got ARM support.

I'm coming from a pure node background and I'm still new to ros so I'm probably missing some basic stuff. I think you're saying that the version on npm works if I use the ros system to define messages and then use gennodejs to generate the javascript for nodes to use to talk ROS using those message types. Is this correct? I was hoping to avoid using `gennodejs` if possible. There doesn't seem to be an example of how to use it. 😕 

I'll give the `kinetic-devel` branch a try again. I wouldn't be surprised if the errors I was seeing are trivially fixed.

chris-smith, 2017-01-06:
You're correct. The version on npm, I believe, has issues with newer versions of node since I wrote it against node 5.

I'd guess you would want to use `kinetic-devel` - if you don't want to use `gennodejs` you can either generate messages through `rosnodejs` by running the `generateMessages` script in tools or you can have messages created on the fly by passing a `onTheFly` argument to the init node call like in the `on_the_fly` example. These do assume your ROS environment is sourced correctly, but if you're able to run rostopic tools you've most likely done so.

There's definitely a lot of documentation that needs to happen too...

chris-smith, 2017-03-24:
The version on `npm` is now up to date with what's on `github` and the latest `kinetic` release so, assuming people are up to date, everything should work cleanly.

closed 2017-03-24 by chris-smith