RethinkRobotics-opensource/rosnodejs #45: Trivial program errors continuously

cinderblock, 2017-01-19:
I'm trying to run this very trivial node program:

```
const rosnodejs = require('rosnodejs');
const std_msgs = rosnodejs.require('std_msgs');

rosnodejs.initNode('my_node')
.then(nh => {
  const pub = nh.advertise('/topic', std_msgs.msg.String);
  const msg = new std_msgs.msg.String();
  msg.data = 'Hello!';
  setInterval(() => {
    pub.publish(msg);
  }, 1000);
});
```

Using the `kinetic-devel` git branch.

It connects to master just fine but then instantly fills the screen with repeated error messages.

```
[INFO] [1484793140.047] (ros): Connected to master at http://localhost:11311!
[ERROR] [1484793140.057] (ros.rosnodejs): Exception trying to advertise service /my_node/get_loggers
[WARN] [1484793140.058] (ros): Unable to setup ros logging services
[ERROR] [1484793140.063] (ros.rosnodejs): Error when publishing message on topic /rosout: TypeError: MessageClass.getMessageSize is not a function
    at Object.serializeMessage (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/tcpros_utils.js:264:34)
    at msgQueue.forEach (/home/cameron/ros/node-test/node_modules/rosnodejs/lib/Publisher.js:168:43)
    at Array.forEach (native)
    at Publisher._handleMsgQueue (/home/cameron/ros/node-test/node_modules/rosnodejs/lib/Publisher.js:162:14)
    at ClientQueue.handleClientMessages (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/GlobalSpinner.js:45:22)
    at GlobalSpinner._handleQueue (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/GlobalSpinner.js:203:24)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
[ERROR] [1484793140.063] (ros.rosnodejs): Error when publishing message on topic /rosout: TypeError:
...
```

I'm a little new to ROS still. The first error & warning looks like it's just that I don't have a ROS logger running.

I'm at a loss as to what the other errors are being caused by. Maybe that would be fixed by logger stuff. I haven't had the time to try it yet.

Any insight would be greatly appreciated.

chris-smith, 2017-01-19:
Sounds to me like it's still pulling old `gennodejs` messages... have you tried generating them through `rosnodejs` or doing it all on the fly? It should just be one argument in the `initNode` call.

```
rosnodejs.initNode('my_node', {onTheFly: true})
.then(() => { ... });
```

Can you also include your `ROS_PACKAGE_PATH` and `CMAKE_PREFIX_PATH` environment variables?

cinderblock, 2017-01-20:
Thanks for the pointers.

I am not using `gennodejs` at all. I'm new to ROS and haven't really touched the message format stuff yet. It is not clear to me how to even use `gennodejs`. I don't have any messages defined which is why I'm assuming that `rosnodejs` would auto generate the needed messages.

I tried adding `{onTheFly: true}` option and that helped some. We fix the initial errors and it no longer instantly fills the terminal with errors. Now they are only printed once every second:

```
[INFO] [1484953795.722] (ros): Connected to master at http://localhost:11311!
[ERROR] [1484953796.732] (ros.rosnodejs): Error when publishing message on topic /topic: TypeError: MessageClass.getMessageSize is not a function
    at Object.serializeMessage (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/tcpros_utils.js:264:34)
    at msgQueue.forEach (/home/cameron/ros/node-test/node_modules/rosnodejs/lib/Publisher.js:168:43)
    at Array.forEach (native)
    at Publisher._handleMsgQueue (/home/cameron/ros/node-test/node_modules/rosnodejs/lib/Publisher.js:162:14)
    at ClientQueue.handleClientMessages (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/GlobalSpinner.js:45:22)
    at GlobalSpinner._handleQueue (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/GlobalSpinner.js:203:24)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
[ERROR] [1484953797.732] (ros.rosnodejs): Error when publishing message on topic /topic: TypeError: MessageClass.getMessageSize is not a function
    at Object.serializeMessage (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/tcpros_utils.js:264:34)
    at msgQueue.forEach (/home/cameron/ros/node-test/node_modules/rosnodejs/lib/Publisher.js:168:43)
    at Array.forEach (native)
    at Publisher._handleMsgQueue (/home/cameron/ros/node-test/node_modules/rosnodejs/lib/Publisher.js:162:14)
    at ClientQueue.handleClientMessages (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/GlobalSpinner.js:45:22)
    at GlobalSpinner._handleQueue (/home/cameron/ros/node-test/node_modules/rosnodejs/utils/GlobalSpinner.js:203:24)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
[...]
```

chfritz, 2017-01-21:
Seems like it still can't find the `.msg` definitions of ROS. Can you post your `CMAKE_PREFIX_PATH` environment variable? Also, just for reference, what version of nodejs are you running?

cinderblock, 2017-01-21:
```
cameron@AuntPolly:~/ros/node-test$ echo $ROS_PACKAGE_PATH
/opt/ros/kinetic/share
cameron@AuntPolly:~/ros/node-test$ echo $CMAKE_PREFIX_PATH
/opt/ros/kinetic
cameron@AuntPolly:~/ros/node-test$ node -v
v7.4.0
cameron@AuntPolly:~/ros/node-test$ uname -a
Linux AuntPolly 4.4.0-53-generic #74-Ubuntu SMP Fri Dec 2 15:59:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

chris-smith, 2017-01-21:
Can you try loading the message package after the `initNode` call? I think it's still pulling the generated messages that come from the kinetic install.

```
const rosnodejs = require('rosnodejs');

rosnodejs.initNode('my_node')
.then(() => {
  const std_msgs = require('std_msgs');
  ...
});
```

chfritz, 2017-01-21:
@cinderblock FYI your code works fine for me when using the `onTheFly` option in node 4.7.2.

cinderblock, 2017-01-21:
@chris-smith With your suggested change, but also with `onTheFly` set to `true`:
```
cameron@AuntPolly:~/ros/node-test$ node .
[INFO] [1485039603.413] (ros): Connected to master at http://localhost:11311!
(node:5063) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Cannot find module 'std_msgs'
(node:5063) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Using `const std_msgs = rosnodejs.require('std_msgs');` instead (which I think is what you meant):

```
cameron@AuntPolly:~/ros/node-test$ node .
[INFO] [1485039715.966] (ros): Connected to master at http://localhost:11311!
[ERROR] [1485039715.973] (ros.rosnodejs): Exception trying to advertise topic /topic
(node:5150) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: type.datatype is not a function
(node:5150) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

And still no messages are received by `rostopic echo /topic` 😕

@chfritz I wonder what has changed between the two versions that would have caused this.

chris-smith, 2017-01-21:
whoops I did. It looks like an issue with on the fly code - the generated classes don't include a `datatype` function that is used when you pass in the message class in the `advertise` call. Try passing in a string for the message type instead.
```
const pub = nh.advertise('/topic', 'std_msgs/String');
```

cinderblock, 2017-01-21:
Wooo Hoo! That seems to have fixed it @chris-smith.

This works:

```
const rosnodejs = require('rosnodejs');

rosnodejs.initNode('my_node', {onTheFly: true})
.then(nh => {
  const std_msgs = rosnodejs.require('std_msgs');
  const pub = nh.advertise('/topic', 'std_msgs/String');
  const msg = new std_msgs.msg.String();
  msg.data = 'Hello!';
  setInterval(() => {
    pub.publish(msg);
  }, 1000);
})
.catch(err => {
  console.log(err);
});
```

Now that basic messages work, I'm sure I'll be back for help when I define my own messages.

closed 2017-01-21 by cinderblock