RethinkRobotics-opensource/rosnodejs #53: Takes a long time to initialize the ROS Node

cjds, 2017-04-06:
I'm running this in an electron app, and to initialize the node it takes a good minute. Is this normal behaviour?

  ```
  rosnodejs.initNode('my_node',{onTheFly: true})
    .then(() => {
        let nh = rosnodejs.nh;
        console.log('Initialized');
        paramName = "Hello";
        paramValue = "Goodbty";
        // a whole new parameter-y world
        nh.setParam(paramName, paramValue);
        console.log('Set Param');
        nh.getParam(paramName)
        .then((paramValue) => {
            console.log(paramValue);
        });

    });
```

chris-smith, 2017-04-06:
Hmm.. definitely unexpected. My first guess would be because you're using on the fly messages - depending on your environment it could take a long time to search the entire `ROS_PACKAGE_PATH` - though I've never seen something on the order of a minute. 

Since you're not actually using any ros messages yet (at least, not in the example you posted), you could turn off on the fly messages and see if that improves it.

If it does, to cut the walk from searching in vain, you could try touching a `CATKIN_IGNORE` in directories under the `ROS_PACKAGE_PATH` where you know there aren't any messages - a good place would probably be in your `node_modules` folder.

chris-smith, 2017-04-16:
Hey @cjds - just wanted to check on your status with this issue.

0phoff, 2017-04-19:
I am having exactly the same issue (also using electron).
I am also getting an error when I try to send a service...
It gets inside the call() function, but then the error appears!

Code:
```
  const Trigger = ros.require("std_srvs").srv.Trigger;

  ...

  let calcPerspective = rosnh.serviceClient('/parking/calc_perspective', 'std_srvs/Trigger');
  rosnh.waitForService(calcPerspective.getService(), 2000).then(function(available) {
    if (available)
    {
      (function () {
      let req = new Trigger.Request();
      calcPerspective.call(req).then(function (resp) {
        win.webContents.send("calc-perspective-done");
      });
      })();
    }
    else
    {
      console.log("failed");
    }
  });
```

Error: 
```
Uncaught Exception:
TypeError: _this7._messageHandler.md5sum is not a function
    at Socket.<anonymous> (/media/Data/GoogleDrive/Programs/Ros/ParkingWithRos/src/gui/node_modules/rosnodejs/dist/lib/ServiceClient.js:329:151)
    at Object.onceWrapper (events.js:290:19)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:185:7)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1077:10)
```

chris-smith, 2017-04-19:
Lots of additional questions...

Where is `rosnodejs` living in your electron application? In the front-end or the back-end? How is electron including `rosnodejs`? Are you using `webpack`?

I've successfully built an electron application using `webpack` with `rosnodejs` in the front-end before but I had to use `__non_webpack_require__` to pull it in - it wasn't able to load any of the messages otherwise. I believe there are ways around this but I never got to exploring them. It doesn't sound like you're having this issue though since it doesn't throw on defining Trigger... what version of ROS are you using?

0phoff, 2017-04-19:
Wasn't sure if I needed to start a different issue for this, so sorry if I had to.. 

I am using Ros Kinetic, and rosnodejs v2.0.1
I am using the rosnodejs package in my main.js script, so that means server-side. I am actually emitting events from the html page to the server script, and then sending them through Ros.
I am not using webpack or any other build toolchain, since this is just a really small gui for a robotics school project! 👍 

Thanks for you help!

chris-smith, 2017-04-19:
Can you run `dpkg -l ros-kinetic-gennodejs` and tell me what version you're using?

chris-smith, 2017-04-19:
Also, how are you getting messages? From the kinetic install or on the fly?

0phoff, 2017-04-22:
I am using ros-kinetic-gennodejs v2.0.1-1 and i am using the messages from the kinetic install!

chris-smith, 2017-04-24:
Can you open another issue for the issue you're seeing with `std_srvs/Trigger`? For the slow initialization, can you point your package to use [this branch](https://github.com/chris-smith/rosjs/tree/initTiming)? It adds some minimal extra logging that will hopefully give us a better idea of where the initialization is spending so much time.

As a comparison, running it on my machine against a script that just calls `initNode` gives me
```
Sync: 5.131ms
OnTheFly: 7.802ms
2017-04-24T13:27:03.044-0400 INFO [ros] Connected to master at http://localhost:11311!
CheckMaster: 540.024ms
Logging: 28.168ms
Time: 28.336ms
```

paulbaumgart, 2017-06-27:
I'm seeing the same problem running via electron (back-end). I've narrowed it down to the initial `registerPublisher` XML RPC call (for publishing to `/rosout`), which takes `29486.790ms` on my machine. Will keep digging...

paulbaumgart, 2017-06-27:
It seems that the nodejs built-in `http.request` is very slow for XML-RPC. Using electron's `net.request` instead is about 500x faster.

Here's a monkeypatch that solves the problem for me. Add it somewhere near the top of your electron main process:

```javascript
const http = require('http');
const {net} = require('electron');
http.request = function(options, cb) {
    if (options.port && options.host && !options.hostname) {
        options.hostname = options.host.split(':')[0];
        delete options.host;
    }

    let req = net.request(options, cb);

    if (options.headers) {
        Object.keys(options.headers).forEach(function(key) {
            req.setHeader(key, options.headers[key]);
        });
    }

    return req;
};
```

chris-smith, 2017-06-28:
Interesting! Have you tried running a simple `rosnodejs` node without using Electron and seen a difference in performance? I'd be slightly surprised if the performance of Node.js' built-in `http.request` was that bad in Electron in general... I'm not aware that `xmlrpc` is doing anything strange with `http` either though.

Great to have a workaround for now! Thanks!

chris-smith, 2017-07-03:
What version of `electron` are you using? I can't reproduce this with `electron` 1.6.2 running on 14.04

clynamen, 2017-10-29:
I am having a similar problem with rosnodejs@2.1.1:
If I execute the example script with node, it works correctly. However, if I execute the same script with electron v1.6.15 the node will connect to the ros master, but the callback is not called before 30s. Also, I can strangely force the callback to be called before 30s by resizing the terminal (???) 
I had the same problem with roslibjs, so maybe it is related to a similar library?

`// Require rosnodejs itself
const rosnodejs = require('rosnodejs');
// Requires the std_msgs message package
const std_msgs = rosnodejs.require('std_msgs').msg;

function listener() {
  // Register node with ROS master
  rosnodejs.initNode('/listener_node')
    .then((rosNode) => {
      // Create ROS subscriber on the 'chatter' topic expecting String messages
      console.log("CALLBACK CALLED!")
      let sub = rosNode.subscribe('/chatter', std_msgs.String,
        (data) => { // define callback execution
          rosnodejs.log.info('I heard: [' + data.data + ']');
        }
      );
    });
}

if (require.main === module) {
  // Invoke Main Listener Function
  listener();
}
`

chris-smith, 2017-10-30:
Did you try the workaround listed above? If the issue is with node's native `http` module then `rosnodejs` and `roslibjs` would both be impacted.

rajnunes, 2017-12-01:
Hi im getting an an error that says 
" Unable to include message package std_msgs - SyntaxError: Unexpected token ="
any idea how to resolve this?

chris-smith, 2018-03-07:
Closing this for now due to inactivity and indication that bug is not in `rosnodejs`.

closed 2018-03-07 by chris-smith