RethinkRobotics-opensource/rosnodejs #54: event for roscore exit

prajankya, 2017-04-16:
If roscore is disconnected, can this library fire an event?

if no, then we can have lazy checking of the last communication happened some 20 secs ago, so its not connected..(communication might count everything from messages, services, api calls)

What is your thought on implementing this?

use case:
This library is used in an nodejs application running on same/different system as roscore, and is connected, its publishing some action based commands(for example /cmd_vel) to roscore..

If in doing so, if roscore is crashed/ killed/ disconnected, this lib should fire an event so that the software can stop sending(atleast show status that roscore is disconnected)

chris-smith, 2017-04-16:
We don't maintain an active connection to the master so firing an event might be difficult. Most communication (after setup, assuming you're not creating or destroying publishers, subscribers, or services) doesn't go through the master either so I don't think detecting whether it is down from topic/service activity would work.

I think the best option would be to "ping" the master periodically through `xmlrpc` calls. Currently `xmlrpc` calls that are rejected because the host is down are all retried indefinitely under the assumption that the master isn't running yet but I have been meaning to thread through options to let you specify a maximum number of attempts to connect. That would let you write something like

```
rosnodejs.nh.getMasterUri({maxAttempts: 1})
.catch(() => {
  console.error('Unable to ping to ROS master');
});
```

We generally use the [`getUri`](http://wiki.ros.org/ROS/Master_API#Name_service_and_system_state) method as a ping mechanism.

This could also be added to the library to fire an event, though I'm not immediately sure if we should... a similar mechanism doesn't exist (to my knowledge) in `rospy` or `roscpp`. An unexpected master-shutdown in ROS is also kind of a catastrophic scenario and you can't just bring it back up without restarting all of the associated ROS nodes - though simply "losing" connection would present similarly and might still be what you're looking for from an application standpoint...

If you want to detect a crash of another ROS node that would likely be easier, assuming that node has at least one publisher/subscriber that is always up. If you create a subscriber/publisher to connect to that topic, you can detect `disconnect` events on both and then check on the number of connected clients, e.g.

```
sub = nh.subscribe('/some_node/alive', 'std_msgs/Empty');
sub.on('disconnect', () => {
  if (sub.getNumPublishers() === 0) {
    console.error('some_node connection lost!');
  }
});
```

prajankya, 2017-04-17:
The disconnect of a subscriber method would add an overhead of keeping up a dummy node running.

I guess I can do a subscribe to /rosout (as it would go down if roscore is killed, i guess) will this work ?

But the maximum attempts option would be great.

I understand it takes time as well as radical shift of code, so if possible can you implement it in near future ? I will do some hack till then.

Thank you for your response.

IanTheEngineer, 2017-04-17:
I can confirm that `rospy` and `roscpp` do not maintain this sort of connection with the `rosmaster` after initial configuration of the node / publishers / subscribers. In fact, a `rosmaster` can go down entirely, and the system operate "normally": messages are still sent and recieved. However, in this situation, no new nodes, publishers or subscribers could attach themselves to the network (due to a lack of `rosmaster`).

I like your idea of a maximum attempts to ping the master:
 ```
 rosnodejs.nh.getMasterUri({maxAttempts: 1})
 .catch(() => {
  console.error('Unable to ping to ROS master');
 });
 ```
That seems reasonable, but I agree that a constant ping to the `rosmaster` would add unnecessary overhead.

prajankya, 2017-04-18:
I verified this doing experiment
> In fact, a rosmaster can go down entirely, and the system operate "normally": messages are still sent and recieved.

And in this case, I guess as `roscpp` as well as `rospy` does the same, then so should rosnodejs, its better to be in norms.

Hence I agree with your decisions and will make my code work in this same fasion.

Thank you for your support. Can close the issue.

closed 2017-04-18 by chris-smith