RethinkRobotics-opensource/rosnodejs #55: Catch error for different datatype

prajankya, 2017-04-16:
How to catch error for a subscriber initiated with different type (say std_msgs/UInt16MultiArray) and receives in different type(say std_msgs/String)?

the library catches this error somewhere, but can it emit this event ?

`[ERROR] [1492332455.554] (ros.rosnodejs): topic types do not match: [std_msgs/UInt16MultiArray] vs. [std_msgs/String]`

chris-smith, 2017-04-16:
I think we don't currently emit anything for this because it seems like programmer error rather than a possible runtime occurrence you would want to guard against.

Is there a scenario you have where this isn't the case? Or a reason you'd like the event in addition to the message?

prajankya, 2017-04-17:
I am developing plugin for [node-red](https://nodered.org/) and it is up to the user which datatype he selects.

Consider a use case 

[image]

I will brief about it, assuming you dont know about node-red, if you do, please neglect it,

Nodered is a graphical programming language, focused towads IOT, but can be used for anything.

The left node is a subscriber (taking in form /cmd_vel) and we obviously know its of type geometry_msgs/Twist as shown in the properties of this node (node is block in node-red):

[image]
as you can see, after the user types in the Topic, he can select which datatype package and the datatype to choose( below my code will fetch the exact format of the object it will give out)

The current problem is if the user selects wrong datatype (take for example TwistStamped), there is no error. if from terminal I published with turtle_teleop_key with datatype Twist, the rosnodejs lib will not crash and no method of knowing what went wrong.

So I think I can implement it if I can get to know the datatype from the other side, before setting it from here (if we know the topic, we can get to know the datatype from rostopic info isn't it?). But this method also would go wrong if there is no publisher yet, we only have a subscriber up. So I guess the answer would be that when the rosnodejs lib detects this fault, and logs the error `[ERROR] [1492332455.554] (ros.rosnodejs): topic types do not match: [std_msgs/UInt16MultiArray] vs. [std_msgs/String]` it emits an event so my node can tell the user that the datatype is mismatching.

what are your view on this?

chris-smith, 2017-04-17:
If there is no publisher yet you still wouldn't get this event - the log you're seeing is from the publisher rejecting the subscriber's connection header and if no publisher is running we won't send any connection header that could be rejected. We do emit events on connect/disconnect, so you could do something like
```
const sub = nh.subscribe('/cmd_vel', 'std_msgs/String');
sub.on('connection', () => {
  setSubConnectionCount(sub.getTopic(), sub.getNumPublishers());
});

sub.on('disconnect', () => {
  setSubConnectionCount(sub.getTopic(), sub.getNumPublishers());
});
```
and then display some 'connected' state based on that.

We could maybe emit 'error' events on the subscriber when they send an invalid connection header... There can also be multiple publishers though so if two publishers are on the same topic using different types your subscriber would get data from one of them but not the other. How would you want to handle this in your plugin? In this scenario the subscriber seems to be functioning correctly.

I would be more concerned about emitting 'error' events when publishers reject invalid incoming connections - that's more a case where the publisher is doing the correct thing in response to some client sending it junk. It certainly shouldn't crash...

Emitting some other event also worries me a bit for both of these cases - it feels a little like we're exposing internal logic and making it part of our public API. 

Another way to do this type of thing would also be by inspecting the ROS graph - there is certainly some work we could do to expose this better. By querying the ROS master and then the other nodes in the graph, you can determine which topics and services are running and what data types they're using - this could be something that you check to give the user more feedback about why their subscriber or publisher doesn't have any connections yet.

prajankya, 2017-04-18:
According to what you said
>  There can also be multiple publishers though so if two publishers are on the same topic using different types your subscriber would get data from one of them but not the other. 

I haven't thought of it this way..

I studied the whole scenario from this perspective and decided on silently neglecting the mismatch of headers issue, and yes, 
>  It certainly shouldn't crash...

So the problem is solved by silently neglecting wrong datatype and continue if there is correct datatype..

Thanks for your timely reply and support. You can close this issue.

closed 2017-04-18 by chris-smith