RethinkRobotics-opensource/rosnodejs #56: Can we Broadcast Transform?

prajankya, 2017-04-23:
Can we make a transform_boardcaster in some way ?

http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20broadcaster%20%28C%2B%2B%29

chris-smith, 2017-04-24:
We could add a similar class to `rosnodejs`... though it may make more sense for that to be an external package using `rosnodejs`. [`tf2`](http://wiki.ros.org/tf2) and a number of other tools are generally separate packages built on top of the core ROS communication infrastructure. Building out the suite of tools available to ROS users in `Node.js` would be a great thing though.

A TransformBroadcaster is also just a convenience wrapper around a ROS publisher, so it would be easy to duplicate its functionality on your own.
e.g.
```
class TransformBroadcaster() {
  constructor() {
    this._pub = rosnodejs.nh.advertise('/tf', 'tf2_msgs/TFMessage', {queueSize: 100});
  }

  // insert other class methods
}
```

prajankya, 2017-04-27:
Thank you, I would look into it, For now I am running transform broadcaster in cpp

closed 2017-04-27 by prajankya