RethinkRobotics-opensource/rosnodejs #59: How to handle UInt64 array (and header with timestamp) when publishing things ?

Tutorgaming, 2017-05-28:
I need to publish "mavros_msgs/Mavlink"  from rosnodejs. 
which has an array of UINT64  (Uint64[]) inside the message. 

All I've done is receiving raw data packet as a buffer, split it and assign to the others field.

For receiving and reading, I've followed the code and found that it used "BN.JS"
, I used it to translate and the use them successfully.

but for the UInt64[], I still have no idea how to pack it as ROSMSG. 

Should I pack it as Array of Bignum [<BN:XXXXXXX> , <BN:YYYYYYYY> , .... ] 
or send it as the array of buffer? 

Also, it has a header which requires timestamp.
Again, I've followed the code and found that I can call rosnode.Time to get now() from that util,
is that the correct thing to do?

chris-smith, 2017-05-30:
To create an `uint64` array field you would simply assign an array of bignums to the field. I believe we also accept anything that can be turned into a bignum (`new BN(x)`) when publishing.

To send it as a ros message, you can either specify all the fields yourself and send it as JSON or you can create a Mavlink message by pulling it in through `rosnodejs`.

```
const mavros_msgs = rosnodejs.require('mavros_msgs');
const Mavlink = mavros_msgs.msg.Mavlink;

const pub = rosnodejs.nh.advertise('/my_topic', Mavlink);
pub.publish(new Mavlink());
// or
pub.publish({
  header: {
    stamp: { secs: 0, nsecs: 0 },
    seq: 0,
    frame_id: 'base'
  },
  framing_status: 1,
  // the rest of the fields
});
```

`rosnodejs.Time.now()` will give you the correctly formatted ROS time or sim time though you can also construct it yourself. In general, `rosnodejs` doesn't care if you publish a Message or JSON. It does care that the message is fully formed - if fields (or nested fields) in your message are missing it may fail to publish your message. Creating the message from the class makes this easier as it guarantees a fully formed object.

Tutorgaming, 2017-05-30:
Thank you for you very well inform.  Now it works like a charm ! 

Thanks for sharing us the good lib :)

closed 2017-05-30 by Tutorgaming