RethinkRobotics-opensource/rosnodejs #63: Writing a tf broadcaster and a tf listener

schadocalex, 2017-07-25:
There are subscribers, publishers, parameters, services... but nothing about tf!

http://wiki.ros.org/tf/Tutorials

IanTheEngineer, 2017-07-25:
Right you are, @schadocalex . Similar to `ros_comm` for C++ & Python, this is a ROS communications library written in nodejs. `tf`, `robot_model`, URDF parser and others are ROS tools built on top of `ros_comm` and not required for communications, therefore placed in a separate repository.

Not having the bandwidth to tackle adding those core ROS tools to build on rosnodejs right now, we welcome (and encourage) them as community contributions, but would likely host them in an adjacent repository. If you're interested, I can create a new repo on the @RethinkRobotics-opensource organization for these efforts.

schadocalex, 2017-07-25:
I was using `roslibjs` and `tf` is supported, that's why I thought it could be supported by `rosnodejs`. But I agree with you about `ros_comm` so yes please create the new repo!

I let you close this issue with the link of the new repo, maybe?

IanTheEngineer, 2017-07-25:
Great, thanks for the recommendation, @schadocalex . Created a repo and Issue to track this:
https://github.com/RethinkRobotics-opensource/tf2_nodejs/issues/1

closed 2017-07-25 by IanTheEngineer