RethinkRobotics-opensource/rosnodejs #67: Subscribing to message types not in std_msgs

emielsteerneman, 2017-09-06:
I'm trying to include rosnodejs into a project, which uses messages that are not in 'std_msgs'. These messages are defined in our own packages. I want to subscribe to these custom messages, but I cannot find a way to do so. Plugging in std_msgs.msg.String gives me the following error : 

"[ERROR] [1504707499.036] (ros.rosnodejs): Client [/jsNode] wants topic /robotcommands to have datatype/md5sum [std_msgs/String/992ce8a1687cec8c8bd883ec73ca41d1], but our version has [myproject_msgs/RobotCommand/bd4e3fb1e84585feac0a9174a19b71be]. Dropping connection.
"

code : 
--------------------------------------------------------------------------------
let ros = require('rosnodejs');
let std_msgs = ros.require('std_msgs')
ros.initNode('/jsNode')
.then((rosNode) => {
  rosNode.subscribe('/robotcommands', std_msgs.msg.String,  msg => {
  	console.log(msg)
  })
})
--------------------------------------------------------------------------------

Is there a way to describe to these messages?

emielsteerneman, 2017-09-06:
By accident, I figured out how to do this using the function require('rosnodejs').getAvailableMessagePackages(). Apparently, my messages are defined in workspace/devel/share/gennodejs/ros/myproject_msgs/_index.js

closed 2017-09-06 by emielsteerneman

chris-smith, 2017-09-06:
The [wiki](http://wiki.ros.org/rosnodejs/overview/messages) has more information about generating messages if you need it.