RethinkRobotics-opensource/rosnodejs #70: Get list of all available services

prajankya, 2017-09-27:
Is there any way to get list of all the services, similar to the `rosnodejs.getAvailableMessagePackages()` function ?

I guess `ros_msg_utils` package need to be ported to services for this, can you please help ?

chfritz, 2017-09-27:
Yes, you can see [how in one of the examples](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/examples/turtle.js#L12-L15):

```js
 // get list of existing publishers, subscribers, and services
 rosNode._node._masterApi.getSystemState("/my_node").then((data) => {
    console.log("getSystemState, result", data, data.publishers[0]);
  });
```

prajankya, 2017-09-27:
I dont want to get the running / available services, I want to get all the srv files available (in - short need  `rossrv` and not `rosservice`)

prajankya, 2017-09-27:
By doing something similar to 

```javascript
var x = rosnodejs.getAvailableMessagePackages()
var a=[];
for (var key in x) {
    if (x.hasOwnProperty(key)) {
        a.push(key.trim());
    }
}
console.log(a);
```
I got this,

```json
[
"actionlib",
"actionlib_msgs",
"actionlib_tutorials",
"bond",
"control_msgs",
"control_toolbox",
"controller_manager_msgs",
"costmap_2d",
"diagnostic_msgs",
"dynamic_reconfigure",
"gazebo_msgs",
"geographic_msgs",
"geometry_msgs",
"grid_map_msgs",
"hector_gazebo_plugins",
"hector_mapping",
"laser_assembler",
"map_msgs",
"nav_msgs",
"nodelet",
"octomap_msgs",
"pcl_msgs",
"polled_camera",
"robot_localization",
"rosapi",
"rosauth",
"rosbridge_library",
"roscpp",
"roscpp_tutorials",
"rosgraph_msgs",
"rospy_tutorials",
"rosserial_arduino",
"rosserial_msgs",
"sensor_msgs",
"shape_msgs",
"smach_msgs",
"std_msgs",
"std_srvs",
"stereo_msgs",
"tf",
"tf2_msgs",
"theora_image_transport",
"topic_tools",
"trajectory_msgs",
"turtle_actionlib",
"turtlesim",
"uuid_msgs",
"visualization_msgs"
]
```

but when I run `rossrv packages`, I get 
```bash
control_msgs
control_toolbox
controller_manager_msgs
diagnostic_msgs
dynamic_reconfigure
gazebo_msgs
geographic_msgs
grid_map_msgs
hector_gazebo_plugins
laser_assembler
map_msgs
nav_msgs
nodelet
octomap_msgs
polled_camera
robot_localization
rosapi
rosauth
roscpp
roscpp_tutorials
rospy_tutorials
rosserial_arduino
rosserial_msgs
sensor_msgs
std_srvs
tf
tf2_msgs
topic_tools
turtlesim
```

There are some packages, I suppose, having only services and no messages, hence `getAvailableMessagePackages()` function cant get those, and thats the reason I am asking this.

Thank you for your time anyways, @chfritz

prajankya, 2017-09-27:
After cleaning up my code, for removing empty packages, 
```javascript
var x = rosnodejs.getAvailableMessagePackages()
var a = []
for (var key in x) {
  if (x.hasOwnProperty(key)) {
    key = key.trim()
    var pack = rosnodejs.require(key);
    if (pack) {
      var list = pack.srv;
      if (list && Object.keys(list).length > 0) { //to remove packages with no srv
        a.push(key)
      }
    }
  }
}
```

I got this

```javascript
[
"control_msgs",
"control_toolbox",
"controller_manager_msgs",
"diagnostic_msgs",
"dynamic_reconfigure",
"gazebo_msgs",
"geographic_msgs",
"hector_gazebo_plugins",
"laser_assembler",
"map_msgs",
"nav_msgs",
"nodelet",
"octomap_msgs",
"polled_camera",
"robot_localization",
"rosapi",
"rosauth",
"rosbridge_library",
"roscpp",
"roscpp_tutorials",
"rospy_tutorials",
"rosserial_arduino",
"rosserial_msgs",
"sensor_msgs",
"std_srvs",
"tf",
"tf2_msgs",
"topic_tools",
"turtlesim"
]
```

still comparing to the commandline `rossrv packages`
```bash
control_msgs
control_toolbox
controller_manager_msgs
diagnostic_msgs
dynamic_reconfigure
gazebo_msgs
geographic_msgs
grid_map_msgs
hector_gazebo_plugins
laser_assembler
map_msgs
nav_msgs
nodelet
octomap_msgs
polled_camera
robot_localization
rosapi
rosauth
roscpp
roscpp_tutorials
rospy_tutorials
rosserial_arduino
rosserial_msgs
sensor_msgs
std_srvs
tf
tf2_msgs
topic_tools
turtlesim
```
Few are missing like
`grid_map_msgs`

I am making a GUI software based on rosnodejs, and hence allow the user to select the type of service from a `drop-down list`, and so getting every package is important.

chris-smith, 2017-09-27:
`getAvailableMessagePackages` returns a map with any package that has built messages or services in your workspace or overlaid workspaces. If messages are missing, it likely means that they can't be found where expected in your `CMAKE_PREFIX_PATH`. Why that would be depends on the version of ROS you're using, how you're generating messages, how you've configured your workspaces, etc.

prajankya, 2017-09-28:
Does the function `getAvailableMessagePackages` gives package which has only `service` and no `message`?

chris-smith, 2017-09-28:
Yup - `std_srvs` only has service message definitions, for example.

prajankya, 2017-09-30:
Great @chris-smith  thank you, its working well.!

closed 2017-09-30 by prajankya