RethinkRobotics-opensource/rosnodejs #74: Subscribe error: couldn't find preset "env"

crunchex, 2017-10-08:
```
  rosnodejs.initNode('/my_node', { rosMasterUri: `http://robot:11311` }).then(n => {
    n.subscribe(
      '/global_state', 'std_msgs/msgs/Int8MultiArray',
      (data) => {
        console.log('SUB DATA ', data, data.data)
      },
      { queueSize: 1, throttleMs: 1000 }
    )
  }).catch(e => console.log(e))
```
yields:
```
[INFO] [1507485372.058] (ros): Connected to master at http://robot:11311!
[WARN] [1507485372.062] (ros): Unable to setup ros logging stream
[WARN] [1507485372.066] (ros): Unable to setup ros logging services
[ERROR] [1507485372.251] (ros.rosnodejs): Exception trying to subscribe to topic /global_state
Error: Unable to include message package std_msgs - Error: Couldn't find preset "env" relative to directory "/opt/ros/kinetic/share/gennodejs/ros/std_msgs"
    at Object.loadMessagePackage (/home/mtl/work/emt/node_modules/rosnodejs/dist/utils/message_utils.js:172:13)
    at Object.getHandlerForMsgType (/home/mtl/work/emt/node_modules/rosnodejs/dist/utils/message_utils.js:214:14)
    at NodeHandle.subscribe (/home/mtl/work/emt/node_modules/rosnodejs/dist/lib/NodeHandle.js:126:44)
    at /home/mtl/work/emt/src/main.js:184:9
    at process._tickCallback (internal/process/next_tick.js:103:7)
```
I can't seem to trace the error referring to "preset 'env'" any farther back then NodeHandle's subscribe function. But all the generated messages appear to correctly be in the path `/opt/ros/kinetic/share/gennodejs/ros/std_msgs/msg/` so I'm wondering if I'm misusing the API somehow? `std_msgs/msgs/Int8MultiArray`  vs `std_msgs/Int8MultiArray` doesn't seem to make a difference.

chfritz, 2017-10-08:
Replace
```js
      '/global_state', 'std_msgs/msgs/Int8MultiArray',
```
with just
```js
      '/global_state', 'std_msgs/Int8MultiArray',
```

crunchex, 2017-10-09:
Actually, I'm seeing some similar issues being created in other projects and it looks like the common thread is babel-preset-env. I'll report back if going in that direction fixes my problem.

chris-smith, 2017-11-10:
Hey @crunchex - can you confirm that `babel` was the cause of this?

chris-smith, 2017-11-27:
going to close this as an issue with `babel` - feel free to reopen if evidence indicates otherwise.

closed 2017-11-27 by chris-smith