RethinkRobotics-opensource/rosnodejs #75: serviceClient.call not receive response callback

cbsim77, 2017-10-30:
run code : 
var req = rosnodejs.require('srv').Request();
								req.city = "city111";
serviceClient.call(req, (resp) => { console.log(resp); });

but can't received response data

SericeClient.js 158 line

key: 'call',
value: function call(request, resolve, reject) {
var _this2 = this;
return new Promise(function (resolve, reject) { 
var newCall = new ServiceCall(request, resolve, reject);

i change 

key: 'call',
value: function call(request, resolve, reject) {
var _this2 = this;
var _resolve = resolve;
return new Promise(function (resolve, reject) {
if(_resolve != undefined) resolve = _resolve;
var newCall = new ServiceCall(request, resolve, reject);

i wonder that is ok

chris-smith, 2017-10-30:
Which version of `rosnodejs` are you using? The `call` function in ServiceClient doesn't currently look like the one you mention.

I'm also curious about your test code - it doesn't look like it should work. The pattern for calling a service would be
```
const MyService = rosnodejs.require('my_message_package').srv.MyService;
const req = new MyService.Request();
req.city = 'city111';
serviceClient.call(req)
.then((resp) => {
  console.log('%j', resp);
});
```

chris-smith, 2017-10-30:
The documentation on the [ros wiki](http://wiki.ros.org/rosnodejs/overview/services) was outdated and required you to pass in a callback function to the ServiceClient.call function instead of returning a promise - it has been updated.

closed 2017-10-31 by IanTheEngineer