RethinkRobotics-opensource/rosnodejs #78: Package traversal with symlinks in workspace is broken (Indigo)

banerjs, 2017-11-10:
I have a workspace with the following structure:

```
- src
  - CMakeLists.txt
  - packages_folder_1 (symlink)
    - folder_1_msgs
      - CMakeLists.txt
      - package.xml
      - msg
        - MessageDef1.msg
        ...
    - folder_1_other
    ...
  - packages_folder_2 (symlink)
  ...
```

My `ROS_PACKAGE_PATH` points to the `src` directory and the default indigo location.

Running the `rosnodejs.loadAllPackages()` in the above tree does not build `folder_1_msgs`. However, if I copy the `folder_1_msgs` out from `packages_folder_1` and put it directly under `src`, the messages get built.

chris-smith, 2017-11-13:
Hi @banerjs - which version of `rosnodejs` are you using? I can't reproduce this on master or the latest npm versions.

banerjs, 2017-11-13:
Interesting. I'm on `2.1.1`. I'll run some more tests later this evening in a new workspace and get back to you on whether there might be something else in my workspace structure that causes the behaviour I'm seeing.

banerjs, 2017-12-05:
Sorry about the extreme delay in this response. I checked with a new workspace and apparently it's an issue that manifests with symlinks only.

Here's what I did, and perhaps you might be able to spot what's going on? For reference, the repository in question is [vector_v1](https://github.com/StanleyInnovation/vector_v1), and the messages I'm trying to build are in `vector_v1/vector_common/vector_msgs`:

1. Create a new workspace and put a symlink to a parent folder (in my case `vector1`) in the `src` directory. Run `catkin_make`. When complete, source the `devel` directory, then run `node`. In the node REPL, run:
```
r = require('rosnodejs')
r.loadAllPackages()
# Some output of the messages built (vector_msgs IS NOT in this list)
exit()

# For some reason, I need to run exit and run node again to get the messages;
# even waiting for a result from the promise in r.loadAllPackages() doesn't work.
# So anyway, run node, and the commands above, again

r.require('std_msgs').msg
# Output of std_msgs
r.require('vector_msgs').msg
# Cannot read property msg of undefined, ERROR
```

2. Empty the workspace and start again. This time, only symlink `vector_v1/vector_common/vector_msgs` in the `src` directory of the workspace. `catkin_make`, source the workspace, and run node again:
```
r = require('rosnodejs')
r.loadAllPackages()
# Some output of the messages built (vector_msgs IS NOT in this list)
# Same deal with the exit and restart
r.require('std_msgs').msg
# Output of std_msgs
r.require('vector_msgs').msg
# Cannot read property msg of undefined, ERROR
```

3. Empty the workspace. This time, **copy** the folder `vector_v1/vector_common/vector_msgs` into the `src` directory of the workspace. `catkin_make`, source the workspace, and run node:
```
r = require('rosnodejs')
r.loadAllPackages()
# Some output of the messages built (vector_msgs IS in this list)
# Same deal with the exit and restart
r.require('std_msgs').msg
# Output of std_msgs
r.require('vector_msgs').msg
# Output of vector_msgs
```

4. Empty the workspace. This time, copy the **entire** `vector_v1` folder into the `src` directory of the workspace. `catkin_make`, source the workspace, and run node:
```
r = require('rosnodejs')
r.loadAllPackages()
# Some output of the messages built (vector_msgs IS in this list)
# Same deal with the exit and restart
r.require('std_msgs').msg
# Output of std_msgs
r.require('vector_msgs').msg
# Output of vector_msgs
```

Hope that helps and you're able to reproduce that to figure out what's going on.

chris-smith, 2017-12-11:
Hmm... I still can't reproduce this. Can you try the following?

```
const r = require('rosnodejs');
const outputDir = 'some_temporary_directory';
r.loadAllPackages(outputDir, true);
```

And then check `some_temporary_directory` for a `vector_msgs` folder. Possibly also look at the logged output for references to `vector_msgs`.

If you still can't find any references to `vector_msgs` I'd guess that something is silently killing part of the process (walking package tree, generating message spec, etc). First place to look would be `findPackagesInDirectory` in `utils/messageGeneration/packages.js` - you should see it find all of your `vector_` packages and it should also find the messages under `vector_msgs`.

chfritz, 2017-12-11:
Could this be the same issue as what @mjyc fixed in https://github.com/RethinkRobotics-opensource/rosnodejs/pull/76 ? I believe that has not yet been published in an npm package (not in 2.1.1).

banerjs, 2017-12-11:
@chris-smith Here's a gist with two output logs from the commands that you mentioned: https://gist.github.com/banerjs/cbfb2b6f7e90377e762d9a58acd7bfc8. `SYMLINK.log` does not have the missing messages, while `NO_SYMLINK.log` does.

I won't have time to inspect the package traversal until next week at the latest, but thanks for the pointer! I'll be sure to update this thread if I find anything when I do.

chris-smith, 2018-10-22:
Closing for now due to lack of activity (and no additional reports of the issue). Feel free to re-open if its still a problem.

closed 2018-10-22 by chris-smith