RethinkRobotics-opensource/rosnodejs #79: [ERROR] [1512132643.557] (ros.rosnodejs): Exception trying to subscribe to topic /chatter

rajnunes, 2017-12-01:
@IanTheEngineer Ive tried executing the same code that is available and I'm getting  this error.

at one point i got an error saying that the package std_msgs was not defined, I've also tried reinstalling rosnodejs and npm just incase there were any problems while installing

chris-smith, 2017-12-01:
What is the exception you're seeing? Is it different than the `std_msgs` one? What code that's available are you referring to?

Also - `rosnodejs` version, ROS version, OS, how are you generating messages?

chris-smith, 2018-01-10:
From the current information this doesn't seem like a bug I'm closing for now - feel free to re-open if you're still experiencing the issue or think otherwise.

closed 2018-01-10 by chris-smith