RethinkRobotics-opensource/rosnodejs #82: Remapping is not working

skkeem, 2018-02-05:
Hi.

I am trying to use rosnodejs for my project, but it is ignoring the remappings set by the command-line arguments or roslaunch files.

I have a node like this
`rosnodejs.initNode('foo').then(....);`
and want to rename it with roslauch like this
`<node pkg="..." name="foo" type="...">
  <remap from="foo" to="bar"/>
</node>`
, but `rosnode list` is just showing /foo.

Is there a quick way to fix this problem?

I am using the version 2.1.1.

Thank you.

chris-smith, 2018-02-05:
`rosnodejs` doesn't currently support remapping, though it's been something I've wanted to add for a while. I imagine it would be fairly simple to include topic remapping calls in `NodeHandle._resolve`.  I'm not sure if node remapping is supported through [roslaunch](https://answers.ros.org/question/198873/how-to-changeremap-ros-node-name-in-launch-file/), though it looks like [command line arguments](http://wiki.ros.org/Remapping%20Arguments#Special_keys) should allow it.

Feel free to make a pull request if you're interested, otherwise I'll try to get it done at some point. Likely relatively soon now that somebody has expressed interest...

Relevant source code for `roscpp` would likely be [here](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/src/libros/node_handle.cpp) and [here](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/src/libros/names.cpp).

skkeem, 2018-02-06:
Alright.
I'll start working on this although I can't guarantee it will be my top priority.
Thanks.

PR #89 (2018-08-02): Add Initial Remapping Support
Resolves #82
-Adds the ability to remap topics, services, and parameters
 through the command line. Also supports special remapping keys
 for node name, ip, host, master and namespace.
-Adds tests for remapping
-Adds tests for some new node initialization options.
closed 2018-08-02 by chris-smith