RethinkRobotics-opensource/rosnodejs #84: Error Installing on Ubuntu 16.04 - ROS Kinetic - "Cannot read property 'Split' of undefined !

shreks7, 2018-02-10:
Getting the following error  - 

```
TypeError: Cannot read property 'split' of undefined
    at Object.<anonymous> (/home/project/electron/project/app/node_modules/rosnodejs/dist/ros_msg_utils/lib/message_cache.js:24:35)
    at Object.<anonymous> (/home/project/electron/project/app/node_modules/rosnodejs/dist/ros_msg_utils/lib/message_cache.js:65:3)
    at Module._compile (module.js:569:30)
    at Module._extensions..js (module.js:580:10)
    at Object.require.extensions.(anonymous function) [as .js] (/home/project/electron/project/node_modules/babel-register/lib/node.js:152:7)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
```

Node - v9.5.0
npm - v5.6.0
ROS Kinetic

shreks7, 2018-02-10:
Sorry my bad - Didnt source setup.bash !

closed 2018-02-10 by shreks7

surfertas, 2018-05-11:
Where did you go to source setup.bash? I am running into same error.

surfertas, 2018-05-11:
I am encountering same issue.
Where did you source setup.bash?

shreks7, 2018-05-11:
After you generate the messages, using` catkin_make`, don't forget to `source devel/setup.bash` before running your node app.

surfertas, 2018-05-12:
Thanks! FMG, are you placing your app inside a catkin directory? (e.g rosnodejs_ws/src/app)