RethinkRobotics-opensource/rosnodejs #85: Standard messages are not built

bgromov, 2018-03-05:
I followed the instruction on [rosnodejs_examples](https://github.com/RethinkRobotics-opensource/rosnodejs_examples), but could not run neither of the nodes. The error is:

```
/home/user/rosnodejs_ws/src/rosnodejs_examples/scripts/talker.js:28
const std_msgs = rosnodejs.require('std_msgs').msg;
                                              ^

TypeError: Cannot read property 'msg' of undefined
    at Object.<anonymous> (/home/user/rosnodejs_ws/src/rosnodejs_examples/scripts/talker.js:28:47)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Function.Module.runMain (module.js:442:10)
    at startup (node.js:136:18)
    at node.js:966:3
```

So, apparently the standard messages were not built. In the dev-space under `gennodejs` I only see the packages local to the workspace.

I tried to add `rosnodejs.loadAllPackages()` before loading `std_msgs`, but it made no difference.

My configuration is:

 - Ubuntu 16.04
 - ROS Kinetic
 - nodejs v4.2.6

Any clue what could be wrong?

chris-smith, 2018-03-05:
What are your `CMAKE_PREFIX_PATH` and `ROS_PACKAGE_PATH` environment variables?

bgromov, 2018-03-13:
Sorry for the delay.

I reduced the setup to a minimum (still doesn't work). There is just ROS Desktop Full workspace at `/opt/ros/kinetic/` and a single catkin overlay workspace at `/home/user/rosnodejs_ws/`. Here is the environment:

```shell
CMAKE_PREFIX_PATH=/home/user/rosnodejs_ws/devel:/opt/ros/kinetic
ROS_PACKAGE_PATH=/home/user/rosnodejs_ws/src/rosnodejs_examples:/opt/ros/kinetic/share
```

chris-smith, 2018-03-14:
I'm assuming this issue occurs for other message packages too (`geometry_msgs`, `tf2_msgs`, etc) in addition to `std_msgs`?

It seems like the issue might be the version of Node that you're using - the generated message code wasn't written to target older versions of Node and the transpiling was added to specifically support Node v4.5 (#40). I'm not familiar with the changes between those versions, but trying to load individual message files in Node 4.2.6 ( just through built-in `require`, not `rosnodejs` ) throws a SyntaxError.

bgromov, 2018-03-20:
I managed to install v4.5.0 with `nvm` (Ubuntu's `apt-get` installs the latest version only), but the result is the same. What is confusing it is completely silent about missed requirements.

Is there a way to make it show debug info and trace why does `rosenodejs.require` fail to generate the messages? Can I force it somehow?

chris-smith, 2018-03-20:
Ah. I think @chfritz is able to use Node v4.5 because he's generating messages on the fly instead of reading them from disk. I would imagine the pre-generated messages won't work in versions of Node.js older than v6.

I'm not sure why `rosnodejs` swallows any errors that happen when requiring message packages (which is to say, it seems like a bug), but you can see the error if you try to require the actual message files from their location on disk.

The relevant function is `requireMsgPackage` in `src/utils/message_utils.js` if you want to open a pull request to rethrow the error.

chfritz, 2018-03-20:
The package.json contains a `prepublish` script that compiles the code using `babel --preset es2015` and generates the `dist/` folder. The main is actually defined as `"main": "dist/index.js"`, i.e., as the babeled version. So you should be able to use node v4.5 just fine.

When are you calling
```
const std_msgs = rosnodejs.require('std_msgs').msg;
```
inside the initNode callback [like in the example](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/examples/turtle.js#L20), or before? If before then that's the problem. The packages have not been loaded yet at that point.

chris-smith, 2018-03-28:
I believe #94 and the referenced README change should address this.

closed 2018-03-28 by chris-smith