RethinkRobotics-opensource/rosnodejs #88: Contents of action goal not exposed?

jstnhuang, 2018-03-21:
If we want to write an action server, how do we get the contents of the goal message? It looks like when an ActionServer constructs a GoalHandle, it only provides a goal ID, but the contents of the goal are not saved anywhere.

chfritz, 2018-03-21:
Hi Justin! Nice to "meet" you this way.

I'm not up to speed on the new ActionServer, I think it's still somewhat early in the development of it. But you can always fall back to just subscribing (goal and cancel) and publishing (result, feedback, and status) messages on the five action topics, because that's all action servers really do.

chfritz, 2018-03-21:
Actually, looking at [the code](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/lib/ActionServerInterface.js#L93), can't you just listen for 'goal' events on the ActionServerInterface?

jstnhuang, 2018-03-21:
Thanks Christian, it's nice to meet you, too.

My understanding is that action servers provide added value on top of publishing to the 5 topics, mostly bookkeeping in terms of the goal IDs and timestamps. I could subscribe to the goal and cancel topics, but without doing the bookkeeping myself, my action server probably won't conform to other action client implementations.

Handling ActionServerInterface's goal event gives me the contents of the goal message, but it doesn't create a GoalHandle.

I'm personally content to work around this issue by just creating a fork and making a small change. However, I'm curious what the long-term plan is for action servers. Does someone already plan on more fully implementing action servers? Or is it something that's awaiting outside contributions?

chris-smith, 2018-03-21:
All very true. The `ActionServerInterface` makes minimal action servers possible, but misses a lot of the details.

Giving access to the actual goal also seems to be something I missed in trying to parse the guts of actionlib's source code...  feel free to make a pull request if you have the opportunity. I haven't had time to come back to it since the initial work.

It's probably also worth mentioning that we recently (yesterday, actually) pulled the actions in `rosnodejs` into a new repo at [`actionlib_nodejs`](https://github.com/rethinkrobotics-opensource/actionlib_nodejs). The main reason for doing so is to allow different JS client libraries (`rclnodejs`, for example) to share a common actionlib backend rather than having to reimplement it on their own. There hasn't been any work there beyond pulling the `ActionServer` and `GoalHandle` classes into the repo. At some point active development on actions will hopefully move there, and how exactly client libraries and the separate `actionlib` repo interact is still up in the air. So, yes, we are planning to finish implementing actions, though outside contributions would be very welcome.

While I'm on the subject of "other" repos, we also created a [`tf2_nodejs`](https://github.com/RethinkRobotics-opensource/tf2_nodejs) repo a little while ago. Also still a work-in-progress. And would also benefit from outside help if anybody has time.

PR #92 (2018-03-22): Added goal to GoalHandle
Relevant to #88.

This change makes the ActionServer pass the goal message to a GoalHandle, which is stored in the `.goal` field. When GoalHandles are created for cancellations, the goal message is left undefined.

This incorporates changes from #91. I put the changes from #91 into a separate pull request since it's a straightforward typo fix, whereas I'm not sure if this is the best way to fix #88.
closed 2018-03-22 by chris-smith