RethinkRobotics-opensource/rosnodejs #93: Custom messages not found?

renatosamperio, 2018-03-23:
Hi,

Many thanks for your library as it has become very useful. However, recently I had found some issues that perhaps you could help me to clarify.

I am trying to make use of custom defined messages from another module but I receive the error:
`TypeError: Cannot read property 'msg' of undefined`

I am including the custom message module in the **package.xml** as `build_depend` and `exec_depend` and in the **CMakelists.txt** in the parts related to `find_package` and `CATKIN_DEPENDS` of the "catkin_package".

I also tried to load the messages explicitly with `rosnodejs.loadAllPackages();` before calling the `require` of the **custom-made module**, but this only works when I execute the node inside the ROS module.

I installed the **node_modules/** inside the ROS module itself. The file structure looks like this:

>my_ws/
>> custom_messages_module/
>>> msg/
>>>> myMessage.msg

>> error_module/
>>> scripts/
>>>>> node_modules/
>>>>> package-lock.json
>>>>> ros_node_script.js

`user@host: ~/my_ws$ rosrun  error_module ros_node_script.js`
`TypeError: Cannot read property 'msg' of undefined`

The file **ros_node_script.js** would look like this:
`#!/usr/bin/env node`
`const rosnodejs = require('rosnodejs');`
`rosnodejs.loadAllPackages();`
`const custom_messages_module = rosnodejs.require('custom_messages_module').msg;`

Any idea would be very helpful.

Regards,
Renato.

renatosamperio, 2018-03-26:
Dear Chris,

Many thanks for your reply. 

I had changed the **onTheFly** part in the _initNode_ as you suggested and it worked. However, I had noticed that **without** the _onTheFly_ section, if I generate a message module with 1 message it keeps finding the module and using correctly the message. However, whenever I use a module with 57 messages it generates an error:

`TypeError: Cannot read property 'msg' of undefined
    at Object.<anonymous> (/home/sfl/workspace/sample_js_ws/src/js_module/scripts/ros_node_script.js:27:65)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
    at Function.Module.runMain (module.js:676:10)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3`

I had dissected as much as I could the issue in the following repository as we have message calls out of the _initNode_ :
- [sample_js_ws](https://github.com/renatosamperio/sample_js_ws.git)

I had made two modules that generate messages. One module follows existing online examples with 1 message **(custom_messages_module)** and the other one has 57 messages and depends on _geometry_msgs_  **(more_custom_msgs)**. 

I had also created a bash file **(build_script.bash)** that is based in the rosnodejs online examples and shows how I am actually compiling the ROS workspace. At the end, it also executes the sample node.

The sample node calls successfully the message module with 1 message but it fails to call the 57 messages module (out of the _initNode_). I had looked around the issue and try different things (e. g. increase node heap, play around with `loadAllPackages`) but I still don't know why the number of messages could be an issue.

Any feedback would be very welcome.

Regards,

Renato.

chris-smith, 2018-03-26:
From the root of your workspace, what is the output of 
```
$ ls devel/share/gennodejs/ros
```
Also, what is the output of `loadAllPackages()`? And what version of node?

renatosamperio, 2018-03-27:
Dear Chris,

I placed the outputs you requested in the following [PASTEBIN](https://pastebin.com/4X8guAsr).

Let me know if it would not be visible.

Regards,

Renato.

chris-smith, 2018-03-27:
I can't reproduce your issue if I run `loadAllPackages()`. Without running `loadAllPackages()` I'm running into the issue in #85 where an error is being swallowed during the require call - your CMakeLists.txt isn't building Sfl_Sound_Vibe. Try requiring the message package on it's own instead of through `rosnodejs`
```
$ node
> require('./devel/share/gennodejs/ros/more_custom_msgs/_index.js')
```

renatosamperio, 2018-03-27:
Hi Chris,

Thanks a lot for your message. I was able to replicate and solve my issue. 

I had removed from CMakelists.txt some (4) `msg` files that were never deleted, I could track them in:
`.../devel/share/gennodejs/ros/hive_msgs/msg/`

but not in `.../devel/.private/hive_msgs/share/gennodejs/ros/hive_msgs/msg/`.

I needed to debug `message_utils.js` and `message_cache.js` to track the issue. Somethow `catkin` still makes use of any message contained in `/msg` and then `rosnodejs` tries to make use of such files. What is **not funny** is that `rosnodejs` crashes silently without saying anything (e. g. files are missing) but a failure in requiring a message.

Regards,

Renato.

chris-smith, 2018-03-28:
Agreed. I believe this should be resolved by #94 - feel free to reopen if something is still missing.

closed 2018-03-28 by chris-smith