RethinkRobotics-opensource/rosnodejs #95: Handling Action Client result and feedback

renatosamperio, 2018-04-03:
Hi,

I would like to know if I should subscribe manually to get **result** and the **feedback** when using the `actionClientInterface` or if there is another way

Any help would be very helpful.

Regards,

Renato.

chris-smith, 2018-04-03:
The `ActionClientInterface` emits `status`, `feedback`, and `result` events whenever it receives a message on one of those topics.

renatosamperio, 2018-04-04:
Hi Chris, thanks for your quick reply.

I implemented the following [client](https://pastebin.com/LTGxj57d) to work with existing [ROS client example](http://wiki.ros.org/actionlib_tutorials/Tutorials/Writing%20a%20Simple%20Action%20Server%20using%20the%20Execute%20Callback%20%28Python%29) in [Python](https://pastebin.com/BW0iYGLZ). Then, I started to debug the file `dist/lib/ActionClientInterface.js`.

As you can see in my code, I am trying to handle the events, however the object `this._goals` does not has the provided `goalId` from the `status.goal_id.id` message for allowing an event `emit` for **result** nor **status** in the implemented handle function. 

My guess is that because of the implementation of the **status**, that is not checking for a `goadId` existing in the object `this._goals`, I can see the **status** from the server on my program implementation. Also, I can see when the `sendGoal` tries to set a `goalId` _(that later will checked for **status** and **result**)_ in `this._goals` with an empty value.

I wonder if I am missing something to declare?

Regards,

Renato.

renatosamperio, 2018-04-04:
Hi, I deleted my previous comment because I saw that I was using an old version when I installed rosnodejs with `npm install rosnodejs`. It got better when I updated the node with the command `npm install npm@rosnodejs -g.`. However it is still telling me the version is 2.2.0.

chris-smith, 2018-04-04:
Version 2.2.0 is the latest version released to npm.

renatosamperio, 2018-04-05:
Many thanks Chris. 

I think these changes to file `lib/ActionClientInterface.js` would allow the **result** and **feedback** events.

```
  }, {
    key: '_handleFeedback',
    value: function _handleFeedback(msg) {
      this.emit('feedback', msg);
    }
  }, {
    key: '_handleResult',
    value: function _handleResult(msg) {
      var cond = this._goals.hasOwnProperty(goalId);
      var goalId = msg.status.goal_id.id;
      if (this._goals.hasOwnProperty(goalId)) {
        delete this._goals[goalId];
      }
      this.emit('result',** msg);
    }
```

What do you think?

Regards,

Renato.