RethinkRobotics-opensource/rosnodejs #103: Getting the latest version of rosnodejs

renatosamperio, 2018-06-19:
Hi,

I had recently installed rosnodejs as described in your instructions. Basically, just did this:

`npm install rosnodejs`

I am using an ubuntu 16.04 with kinetic system.

After downloading the latest version (**rosnodejs@^2.2.0**) I am not seeing the latest version with a [disabled version of the latching in action lib](https://github.com/RethinkRobotics-opensource/rosnodejs/commit/fa98c2721f719fcfce081633411396f1ee06d48f).

Could you let me know how could I get the latest version of rosnodejs?

Renato.

chris-smith, 2018-06-19:
I'd recommend installing from source. Our `.npmignore/.gitignore` makes installing through npm not work too well...

closed 2018-06-19 by chris-smith

chfritz, 2018-06-19:
@chris-smith isn't the issue simply that the commit @renatosamperio is after was made after the last npm release? In `npm info rosnodejs` I see that the last release was bundled in February:
```
     '2.2.0': '2018-02-19T21:41:44.547Z' },
```
but the latching fix was only [merged on May 3rd](https://github.com/RethinkRobotics-opensource/rosnodejs/pull/97).

chris-smith, 2018-06-19:
Yes that's true - we haven't updated the version of `rosnodejs` on npm in a while.