RethinkRobotics-opensource/rosnodejs #104: Inaccurate message type names generated by ActionClientInterface

jstnhuang, 2018-06-20:
I think the way the type names for an ActionClient are generated is inaccurate. If you create an action client for a type `foo_msgs/Bar` (filename Bar.action), the correct message types are `foo_msgs/BarActionGoal`, `foo_msgs/BarActionFeedback`, etc.

This should be true even if the type name ends with "Action." For example, if the action type is `foo_msgs/BarAction` (filename BarAction.action), the correct message types should be `foo_msgs/BarActionActionGoal`, `foo_msgs/BarActionActionFeedback`, etc.

The ActionServerInterface correctly appends "ActionGoal", "ActionFeedback", etc. to the base name of the action type. However, ActionClientInterface only appends "Goal", "Feedback", etc. The string "Action" is appended beforehand in NodeHandle.js, but only after this check:
```js
if (!type.endsWith('Action')) {
  type += 'Action';
}
```

If you have an action whose type name ends with "Action," this leads the ActionClientInterface to generate type names like `foo_msgs/BarActionGoal` instead of `foo_msgs/BarActionActionGoal`. This leads to md5 matching errors with other nodes, since `foo_msgs/BarActionGoal` is also a valid type.

I think the fix should be simple: remove the three lines quoted above from NodeHandle.js, and change ActionClientInterface to always append "ActionGoal", "ActionFeedback", etc. to the type name.

Please let me know what you think. Is this just an edge case that should be fixed, or is it likely that lots of people depend on this behavior? If we want to fix it, I'd be happy to make the pull request.

chris-smith, 2018-06-20:
Going off memory, users used to be required to include the extra "Action" in the actionType field when creating `ActionClients`. So if your action was `foo_msgs/BarAction`, you had to put `foo_msgs/BarActionAction`. I added the line in `NodeHandle.js` as a way to provide backwards support for people still doing that - if you'd been using `foo_msgs/BarAction` it would keep working, but if you started new stuff, you could just do `foo_msgs/Bar`. It fails on actions whose messages _actually_ end in Action, but that probably seemed like a small enough case at the time.

Whatever the reason, at this point I think it's ok for us to do the actual fix ( always include 'Action' in `ActionClientInterface` and remove the check from `NodeHandle` ) - I'm expecting the next release of `rosnodejs` to need a major version bump.

louisgv, 2018-06-22:
@chris-smith is this being worked on? Any chance I can assist with taking a whack at it? 
Disclaimer: I'm a student who's taking an opensource class and my goal is to make contribution as best as I could. I've been to ROS and have been observing this repo for a while, and have been playing around with the code base.

chris-smith, 2018-06-25:
I haven't done anything here - not sure if @jstnhuang has started any work yet though.

PR #106 (2018-06-28): Changed ActionClientInterface to expect action type names without extra "Action" appended
Fixes #104 

Changes the ActionClientInterface such that the type name is always of the form: `my_msgs/DoSomething`, where the action was defined in a file `DoSomething.action` in the package `my_msgs`.

**This is a breaking change!**
- If you previously provided the name with the word "Action" at the end, this will no longer work:
  ```diff
  - const ac = nh.actionClientInterface('/turtle_shape', 'turtle_actionlib/ShapeAction');
  ```
  You must remove the extra "Action" from the action type name:
  ```d [... cut]
closed 2018-06-28 by chris-smith