RethinkRobotics-opensource/rosnodejs #117: bad deserialize perfomance with image data array

panshengjie, 2018-10-22:
I've check the source code, In 

> deserializeInnerMessage (message.js)

reading buildin-type array one-by-one using buffer.readUInt8 ,eating-up the CPU.

chris-smith, 2018-10-22:
I believe you've stumbled on a case where we didn't update the on-the-fly message code to work the same way as the pre-generated messages. The generated messages specifically deal with uint8 arrays by slicing out a section of the overall message buffer rather than iterating over the length of the array.

panshengjie, 2018-11-12:
@chfritz  I was subscirbed to a image topic, with type sensor_msgs/Image, and found a significant rise on cpu usage

panshengjie, 2020-07-01:
for uint8 array, I've forced  converting buffer to Uint8Array . BUT this is not a general resolution, may lead to compatibility issues