RethinkRobotics-opensource/rosnodejs #118: bad performance when deserialize long buildin-type array

panshengjie, 2018-10-22:
rosnodejs deserialize buildin-type array data oneByone using deserializePrimitive(),   can we use typeArray instead?

chris-smith, 2018-10-22:
As far as I'm aware. The uint8 arrays for pre-generated messages are already specially deserialized so we'd just be extending the case.

panshengjie, 2018-11-12:
@chfritz  not the same.  I thought we should use typeArray everywhere  supported,  great performance improve you know.  But this may lead to compatibility issues.

panshengjie, 2018-11-15:
it's quite sure that, TypedArray is not a  superset of normal Arrays, which is introduced for binary operation that nodejs lacks. Replace all arrays with typeArray is impossiable, especially object-arrays.
 
a good choice I think about is introduce a flag to let programmer manually handle heavy binary datas like image/video/sound

besides, heavy object-arrays are still tough , typically laserscan/pointclouds in ROS.

chris-smith, 2018-11-27:
I started looking into doing this but discovered that the TypedArray views require that the byte offset be a multiple of the type size (see: [StackOverflow](https://stackoverflow.com/questions/15417310/why-typed-array-constructors-require-offset-to-be-multiple-of-underlying-type-si)). In specific circumstances (e.g for certain message definitions) it can be guaranteed that this behavior will hold, but turning on TypedArray deserialization for specific topics or messages and not others would probably be a more invasive change than what I'd been imagining...

The general way around the byte-offset issue for ArrayBuffers seems to be using a DataView, but the API there looks like we'd end up with basically the same code we currently have now using Node.js Buffers (which may just be a DataView at this point??).

I'll look into it a bit more... I could be missing something in TypedArrays that would get around this or overcomplicating the per-topic deserialization case.

trusktr, 2020-06-04:
Any plans? This would be a substantial performance improvement. Node.js supports `DataView` for a long time now. I would rather be given a `DataView` and do my own reading of the data than receive an `Array` and have no control of the performance.

trusktr, 2020-06-05:
On my end, I am doing the following, which is similar to what rosnodejs could do. Note in my example it doesn't make rosnodejs any more efficient, it only makes my code more efficient, but it shows how a single `ArrayBuffer` could be re-used without allocating new memory:

```js
      let imageData: Uint8Array
      let imageBlob: Blob
      let imageUrl: string
      let imageBlobParts: BlobPart[] = []
      const img = document.getElementById('cameraImage') as HTMLImageElement

      let buffer = new ArrayBuffer(INITIAL_SIZE)

      const callback = (msg) => {
        if (imageUrl) URL.revokeObjectURL(imageUrl)

        const dataLength = msg.data.length

        if (dataLength > buffer.byteLength) buffer = new ArrayBuffer(dataLength + 10000)

        // TODO make an object pool of Uint8Array objects so we don't have to
        // re-allocate them when we encounter an image that had the same size as
        // a previous image.
        imageData = new Uint8Array(buffer, 0, dataLength)
        imageData.set(msg.data)

        imageBlobParts[0] = imageData // re-use objects as much as possible!
        imageBlob = new Blob(imageBlobParts, {type: 'image/jpeg'})
        imageUrl = URL.createObjectURL(imageBlob)

        img.src = imageUrl // render the image
      }
```

As you can tell there, there is a single spot in memory (a single `ArrayBuffer`), and each time I receive a message, I am re-using the memory and overwriting the values.

If I do the following, the performance is much worse (400% slower):

```js
      let imageData: Uint8Array
      let imageBlob: Blob
      let imageUrl: string
      let imageBlobParts: BlobPart[] = []
      const img = document.getElementById('cameraImage') as HTMLImageElement

      const callback = (msg) => {
        if (imageUrl) URL.revokeObjectURL(imageUrl)

        imageData = new Uint8Array(msg.data) // new one every time, very bad
        imageData.set(msg.data)

        imageBlobParts[0] = imageData // re-use objects as much as possible!
        imageBlob = new Blob(imageBlobParts, {type: 'image/jpeg'})
        imageUrl = URL.createObjectURL(imageBlob)

        img.src = imageUrl // render the image
      }
```

In this last example we are not re-using memory, and the callback is 4x slower.

I think it would be great if rosnodejs would allow us to re-use memory and greatly improve performance.

chris-smith, 2020-06-08:
Excluding on-the-fly messages, rosnodejs already does not allocate additional memory for uint8 arrays. For most other types of message, I think the current api does what users expect, though point clouds and some other data structures may have issues.

The easiest way around this is to allow users to do custom deserialization from the buffer. A longer term solution could be another deserialization option that wraps DataView.

panshengjie, 2020-06-30:
@trusktr I did't quite look into your problem, but seems you are passing image throught rosnodejs to web.  The way I work around is hack the lib, force use typeArray for uint8/int8 array:
` src/utils/messageGeneration/messages.js`
```javascript
 if (arrayType === "int8") {
        array = new Int8Array(buffer.slice(bufferOffset[0], bufferOffset[0] + arraySize))
        bufferOffset[0] += arraySize
      } else if (arrayType === "uint8") {
        array = new Uint8Array(buffer.slice(bufferOffset[0], bufferOffset[0] + arraySize))
        bufferOffset[0] += arraySize
      } else {
````

you can use my version of rosnodejs by npm:  npm i --save @panshengjie/rosnodejs

my approach focus on cpu usage,  mem is not my focus. 

besides, if you stream image to web realtime, you must use a lazy strategy: let the client request for image one by one. same to pointcloud

panshengjie, 2020-06-30:
forgot to say, my approach works with **onTheFly** turned on

panshengjie, 2020-06-30:
I will PR tomorrow with my approach,  5x faster on topic listen