RethinkRobotics-opensource/rosnodejs #119: Unicode characters in strings causes fatal error (using on-the-fly messages)

chfritz, 2018-11-07:
At least when using on-the-fly message definitions, when publishing messages that include strings with Unicode characters this fatal error occurs:

```
RangeError: Index out of range
    at checkInt (buffer.js:1187:11)
    at Buffer.writeUInt32LE (buffer.js:1265:5)
    at /home/cfritz/tmp/out_of_bounds/node_modules/rosnodejs/dist/utils/messageGeneration/messages.js:545:16
    at Array.forEach (<anonymous>)
    at serializeInnerMessage (/home/cfritz/tmp/out_of_bounds/node_modules/rosnodejs/dist/utils/messageGeneration/messages.js:540:15)
    at Function.Message.serialize (/home/cfritz/tmp/out_of_bounds/node_modules/rosnodejs/dist/utils/messageGeneration/messages.js:488:5)
    at Object.serializeMessage (/home/cfritz/tmp/out_of_bounds/node_modules/rosnodejs/dist/utils/tcpros_utils.js:195:18)
    at msgQueue.forEach.msg (/home/cfritz/tmp/out_of_bounds/node_modules/rosnodejs/dist/lib/impl/PublisherImpl.js:224:43)
    at Array.forEach (<anonymous>)
    at PublisherImpl._handleMsgQueue (/home/cfritz/tmp/out_of_bounds/node_modules/rosnodejs/dist/lib/impl/PublisherImpl.js:219:16)
```

This is the minimal code I was able to reproduce this with:

```js
const rosnodejs = require('rosnodejs');

rosnodejs
  .initNode('/test', { onTheFly: true })
  .then((rosNode) => {

  const pub = rosNode.advertise('/rosout', 'rosgraph_msgs/Log', {});
  const Log = rosnodejs.require('rosgraph_msgs').msg.Log;

  const msg = '’'; // a unicode character

  pub.publish(new Log({
    level: 1,
    name: rosNode.nodeName,
    msg,
    file: '',
    function: 'test',
    line: 0,
    topics: []
  }));
});
```

chfritz, 2018-11-07:
@chris-smith do you know whether the gennodejs-generated messages handle this more gracefully?

chris-smith, 2018-11-08:
Hmmm... I know we've tested unicode before. From looking at the code it looks like the on-the-fly messages are using the same method as gennodejs does. I'll have to actually test it out to see what's up.

chris-smith, 2018-11-08:
Ah that makes sense. We use the [Buffer.byteLength](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/ros_msg_utils/lib/encoding_utils.js#L29) function to do this.

PR #120 (2018-11-13): Fixes bug in on-the-fly message handling of unicode strings
Fixes #119.

Thanks for the pointer @chris-smith !
commit 63c6e0e4 (2018-11-13): Fixes bug in on-the-fly message handling of unicode strings (#120)

* Fixes bug in on-the-fly message handling of unicode strings

Fixes #119.
closed 2018-11-13 by chris-smith