RethinkRobotics-opensource/rosnodejs #123: ROS Melodic Support

808brick, 2019-02-03:
I hate to bother, as I'm sure you work tirelessly on creating new features in allowing js developers to utilize ROS, but are you at all considering supporting development for ROS melodic? Your ROS package documentation says Kinetic only and if I try to install the module and run a script on melodic it just says rosnodejs is not defined.

chris-smith, 2019-02-04:
`rosnodejs` should work fine in melodic - I didn't keep making new branches for future ROS versions because there hasn't been (as far as I'm aware) any real changes to ROS communications in a while. If Node.js can't find `rosnodejs` after installing (assuming the install worked fine) I'd guess there's something about your directory structure so that `require('rosnodejs')` fails. Maybe [this](https://nodejs.org/api/modules.html#modules_all_together) will help?

808brick, 2019-02-04:
Thank you for the quick reply. You are right, and thanks for the resource. I was requiring rosnodejs and for some reason when I ran ```npm install``` on my project, which should have installed rosnodejs as specified in my package.json file, it didn't, which was throwing the issue. My issue is resolved.