RethinkRobotics-opensource/rosnodejs #131: Question: Can I avoid installing ROS if my robot can directly respond to node commands?

JamesNewton, 2019-08-30:
Apologies in advance if this is a truly stupid question, but:

I'd like to NOT install ROS on my actual robot to save space. If this rosnodejs can receive and respond to ROS messages, can't I just start with it and add the code required to move the robot and return data from the robots sensors and skip ROS entirely? The goal is just for people who use ROS on their PC's or whatever to be able to talk to my robot via the ROS messaging system. 

Again, pardon me for asking this without truly understanding ROS, but perhaps if you answer it, this will help others understand the system as a whole and rosnodejs's place in it.

chfritz, 2019-08-30:
Yes, you can, if you:
- use the `onTheFly` option (see [here](https://github.com/RethinkRobotics-opensource/rosnodejs#generating-messages)), and
- you have the message definition files of the message types you want to use in your `ROS_PACKAGE_PATH` (path pointed to by that env variable).

JamesNewton, 2019-08-30:
Excellent! I've made a note and will investigate this in the near future.

closed 2019-08-30 by JamesNewton

justinscorringe, 2019-09-30:
Hi. I have a similar question, but with the situation reversed. I want to interface with a remote ROS master uri with a Rosnodejs node, without ROS installed on the nodes system. Is this possible or will I require a ros install/ros docker? Thank you!

chfritz, 2019-09-30:
Yes, that's possible using the above mentioned `onTheFly` option, see example. However you will need to copy the message definition files from the robot (e.g., for ROS melodic in `/opt/ros/melodic/share`) to your machine, and set your `ROS_PACKAGE_PATH` path environment variable to the folder you copied it to.

justinscorringe, 2019-10-01:
Ah thank you!  I have followed through with this and copied the share folder over and set the `ROS_PACKAGE_PATH` to this folder. 
When following the turtlesim example, with `{ rosMasterUri : '192.168.5.109:11311' }` (master address on local network) in the initnode function, I get 
`Unable to find CMAKE_PREFIX_PATH environment variable. Did you source setup.bash?` 
I've tried to also copy these over from the melodic folder and source them but have been unsuccessful in my attempts, am I going about this the right way?

chfritz, 2019-10-01:
Ah yes, you should set that env variable as well, and it usually (always?) the same value as `ROS_PACKAGE_PATH`.

justinscorringe, 2019-10-01:
Yes that worked! Things started up and the ros master connected! Just a few new errors in the process;

`[INFO] [1569893430.163] (ros): Connected to master at 192.168.5.109:11311!`
`[WARN] [1569893430.165] (ros): Unable to setup ros logging stream`
`[WARN] [1569893430.165] (ros): Unable to setup ros logging services`
`(node:2103) UnhandledPromiseRejectionWarning: Error: Unable to find message package turtlesim from CMAKE_PREFIX_PATH`

However turtlesim package is definitely in there!

chris-smith, 2019-10-01:
Can you give more of the error stack?

justinscorringe, 2019-10-01:
Heres the rest; sorry I am terrible at formatting this! 

`TypeError: Cannot read property 'match' of undefined
    at Object.getAddressAndPortFromUri (/Users/justinscorringe/node_modules/rosnodejs/dist/utils/network_utils.js:76:27)
    at new MasterApiClient (/Users/justinscorringe/node_modules/rosnodejs/dist/lib/MasterApiClient.js:32:56)
    at new RosNode (/Users/justinscorringe/node_modules/rosnodejs/dist/lib/RosNode.js:71:23)
    at Object.initNode (/Users/justinscorringe/node_modules/rosnodejs/dist/index.js:113:21)
    at listener (/Users/justinscorringe/node_modules/rosnodejs/dist/examples/rosnodejs_examples/scripts/listener.js:32:13)
    at Object.<anonymous> (/Users/justinscorringe/node_modules/rosnodejs/dist/examples/rosnodejs_examples/scripts/listener.js:45:3)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)`

Here is how I have initialised the node in the example;
`rosnodejs.initNode('/listener_node', { rosMasterUri : '192.168.5.109:11311' }, { onTheFly: true })`

Thank you for looking at this!

JamesNewton, 2020-08-11:
Just to document this a bit more:

On the standard install of Kinetic on Ubuntu, the path variables are
ROS_PACKAGE_PATH="/opt/ros/kinetic/share"
CMAKE_PREFIX_PATH="/opt/ros/kinetic"
To emulate those as close as possible, a ros folder is added wherever, and a "share" folder under that. Those can be added permanently with:
````
echo export ROS_PACKAGE_PATH=/whatever/share/>>/etc/environment 
echo export CMAKE_PREFIX_PATH=/whatever/ros/>>/etc/environment 
````
Wherever they point, there needs to be a set of folders under the share folder with message package.xml and msg folder with the .msg files. Something like you find from /opt/ros/kinetic/share/ in a ROS installation. e.g. the std_msgs folder. But watch out for depends which must also be included. e.g. the package.xml for std_msgs lists:
````
<build_depend>message_generation</build_depend>
<run_depend>message_runtime</run_depend>
````
so the message_generation and message_runtime folders must also be included. To support std_msgs and sensor_msgs, I ended up copying in
- geometry_msgs
- message_generation
- message_runtime
- sensor_msgs
- std_msgs

This allows messages with all the standard string, int32, etc... types from std_msgs, as well as JointState (from sensor_msgs), and Accel, Pose, Quaternion, and others from geometry_msgs. It does not support logging, which seems to have a massive list of dependencies. Of course, you can just copy the entire ros share folder to the robot.