RethinkRobotics-opensource/rosnodejs #132: UTF-8 multi-byte character serialization broken in 3.0.0

mvollrath, 2019-10-07:
I'm working on a project using rosnodejs and non-English languages and have come across a silent data loss issue when serializing strings containing multi-byte UTF-8 characters.

This issue is fixed in kinetic-devel, but it has not been released yet.  A 3.0.1 release with the UTF-8 fixes would be greatly appreciated!

mvollrath, 2019-10-07:
I've opened https://github.com/RethinkRobotics-opensource/gennodejs/pull/19 to fix this in gennodejs as well.

chris-smith, 2019-10-07:
sure can do - I'll probably wait for releasing until the fix in `gennodejs` is merged and I can add some unicode tests here.

mvollrath, 2019-10-08:
> sure can do - I'll probably wait for releasing until the fix in `gennodejs` is merged and I can add some unicode tests here.

I've opened #133 with a UTF-8 string test that fails on 3.0.0 and passes on kinetic-devel.

mvollrath, 2019-10-14:
@chris-smith anything else I can do to help make this release?

chris-smith, 2019-10-14:
Nope! Sorry about the delay - doing it now.

chris-smith, 2019-10-14:
3.0.1 should be up - not sure when an official `gennodejs` release can happen, but including it in overlays should work until then (or using `rosnodejs` messages)

closed 2019-10-14 by chris-smith

mvollrath, 2019-10-14:
Thank you :tada: