RethinkRobotics-opensource/rosnodejs #135: xmlrpc dependency on a github repo

rudicus, 2019-11-18:
Hey Chris,

Our buddy Neil is trying to setup npm in our CI. We are having an issue where npm fails because it tries to download a github repo (and Neil doesn't want git installed on our build slaves). We tracked it down to rosnode.js and it's dependency on a custom fork of  xmlrpc (chfritz/node-xmlrpc)

Seems to sort of be related to:
https://github.com/RethinkRobotics-opensource/rosnodejs/issues/127

Is there a reason the node-xmlrpc can't be made into a new NPM package and published?

We want to be be able to do full repeatable builds without depending on external git connectivity. 

Thanks,

Matt

realtime-neil, 2019-11-18:
There's no reason `node-xmlrpc` can't be published to the npm registry. Even this idiot managed to do it: https://www.npmjs.com/package/node-xmlrpc/v/1.3.3

chfritz, 2019-11-18:
> Is there a reason the node-xmlrpc can't be made into a new NPM package and published?

I think you know the answer to that question, and you could create such an npm package and then create a branch of rosnodejs where you depend on that instead.

However I think the right solution would be for https://github.com/baalexander/node-xmlrpc/pull/132 to get merged and a new version of node-xmlrpc to be published. Unfortunately that is not under the control of Chris or myself.

BTW, I thought that npm pulls github dependencies via HTTPs, not ssh/git. So it's hardly any different from other npm registry downloads.

realtime-neil, 2019-11-18:
@chfritz it's different in that I can install `ca-certificates` and expect it to Just Work(tm) without installing `ssh`, collecting host keys, configuring ssh options, and/or installing `git`. Insofar as I know, pulling packages from a npm registry does not require any of those host manipulations.

rudicus, 2019-11-18:
@chfritz - thanks for the response and work on rosnode.js. 

Yes, we have done that. (hence the https://www.npmjs.com/package/node-xmlrpc/v/1.3.3)  I just hate to split the eco system even more than it already is. Every time a fork is made for one tiny two line change I die a little inside ;) 

In the interim we'll two fork both projects and republish ourselves as you mentioned (unless you guys want to switch to the fork of xmlrpc we created)

chris-smith, 2019-11-18:
Yeah I'm not a big fan of forking/republishing `xmlrpc` if we can avoid it. It's unclear to me how much choice we have if we don't want a GitHub dependency inside the `package.json` though. And getting multiple reports about this issue is a stronger indicator that we should actually do something about it...

chris-smith, 2020-09-28:
This was addressed in #159 but has not been released.

closed 2020-09-28 by chris-smith