RethinkRobotics-opensource/rosnodejs #140: npm install

harryman1, 2020-02-04:
been trying to install rosnodejs and I have this error, please indicate how to solve thsi issue

npm install rosnodejs
npm ERR! code ETARGET
npm ERR! notarget No matching version found for rosnodejs@3.5.2.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-04T14_57_27_025Z-debug.log

the log shows as 

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'install', 'rosnodejs' ]
2 info using npm@6.13.6
3 info using node@v13.7.0
4 verbose npm-session b566381c8b40833d
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 200 https://registry.npmjs.org/rosnodejs 281ms
8 silly fetchPackageMetaData error for rosnodejs@3.5.2 No matching version found for rosnodejs@3.5.2.
9 timing stage:rollbackFailedOptional Completed in 1ms
10 timing stage:runTopLevelLifecycles Completed in 1145ms
11 verbose type version
12 verbose stack rosnodejs: No matching version found for rosnodejs@3.5.2.
12 verbose stack     at pickManifest (/usr/local/lib/node_modules/npm/node_modules/npm-pick-manifest/index.js:122:13)
12 verbose stack     at /usr/local/lib/node_modules/npm/node_modules/pacote/lib/fetchers/registry/manifest.js:24:14
12 verbose stack     at tryCatcher (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/util.js:16:23)
12 verbose stack     at Promise._settlePromiseFromHandler (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:517:31)
12 verbose stack     at Promise._settlePromise (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:574:18)
12 verbose stack     at Promise._settlePromise0 (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:619:10)
12 verbose stack     at Promise._settlePromises (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:699:18)
12 verbose stack     at _drainQueueStep (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:138:12)
12 verbose stack     at _drainQueue (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:131:9)
12 verbose stack     at Async._drainQueues (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:147:5)
12 verbose stack     at Immediate.Async.drainQueues [as _onImmediate] (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:17:14)
12 verbose stack     at processImmediate (internal/timers.js:456:21)
13 verbose cwd /home/Web_Server
14 verbose Linux 4.9.140-tegra
15 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "rosnodejs"
16 verbose node v13.7.0
17 verbose npm  v6.13.6
18 error code ETARGET
19 error notarget No matching version found for rosnodejs@3.5.2.
20 error notarget In most cases you or one of your dependencies are requesting
20 error notarget a package version that doesn't exist.
21 verbose exit [ 1, true ]

chfritz, 2020-02-04:
> No matching version found for rosnodejs@3.5.2.

It seems you have an error in your package.json, specifying a version that doesn't exist. The latest is 3.0.2.

chris-smith, 2020-02-06:
I would agree with @chfritz - closing for now assuming that's the case.

closed 2020-02-06 by chris-smith