RethinkRobotics-opensource/rosnodejs #146: Cannot import custom messages

jmachuca77, 2020-04-17:
When trying to import some custom messages I get an error:

```
my_node.js: Identifier 'my_custom_message' has already been declared
```
when checking the generated files in `my_ws/devel/share/gennodejs/ros` I see the files are generated for my node, but they have duplicated entries in _index.js 

```
"use strict";

let my_custom_message1 = require('./my_custom_message1.js');
let my_custom_message2 = require('./my_custom_message2.js');
let my_custom_message1 = require('./my_custom_message1.js');
let my_custom_message2 = require('./my_custom_message2.js');

module.exports = {
  my_custom_message1: my_custom_message1,
  my_custom_message2: my_custom_message2,
  my_custom_message1: my_custom_message1,
  my_custom_message2: my_custom_message2,
};

```

Any idea what could be causing this? Manually removing the duplicates fixes the issue but I would rather not have to do this every time I do a build.

chris-smith, 2020-04-17:
Is there any chance you have multiple files with these names in your package? Something like
```
my_pkg/A/my_custom_message1.msg
my_pkg/A/my_custom_message2.msg
my_pkg/B/my_custom_message1.msg
my_pkg/B/my_custom_message2.msg
```

This also appears to be an issue with [`gennodejs`](https://github.com/RethinkRobotics-opensource/gennodejs).

jmachuca77, 2020-04-17:
No, the structure of the package is:

my_pkg
   |--- cfg
   |         |--- my_custom_package.cfg (dynamic reconfig file)
   |
   |--- include
   |         |--- my_custom_package.h
   |
   |--- msg
   |         |--- my_custom_message1.msg
   |         |--- my_custom_message1.msg
   |
   |--- launch
   |         |--- my_custom_package.launch
   |   
   |--- src
             |--- my_package_c_code.cpp
             |--- my_package_python.py

I will open an issue in gennodejs, you are right this seems to be more of an issue with that package.

closed 2020-04-17 by jmachuca77