RethinkRobotics-opensource/rosnodejs #147: String serialization failed if it includes unicode character over 2 bytes

lesomnus, 2020-04-18:
```js
let pub = rosNode.advertise('/chatter', std_msgs.String);
const msg = new std_msgs.String();
msg.data = 'Hello, 世界'
pub.publish(msg);
```
Above script will fail because "世界" is length in 6 (not 2).
Currently, in my point of view, `rosnodejs` asserts (throws) if the serialized byte length is different from the expected length of the before serialization which is simply calculated by `data.length` (I don't know exactly about this).

e.g.
Using `String.length`,
```
'世界'.length === 2
'world'.length === 5
```

but in the byte length,
```
const encoder = new TextEncoder();
encoder.encode('世界').length === 6 // attention
encoder.encode('world').length === 5
```

chris-smith, 2020-04-18:
I can't reproduce this. What versions of ros/rosnodejs/gennodejs and what message generation are you using?

lesomnus, 2020-04-19:
it's weird, it only fails if there are more than 4 letters of unicode.
```js
let pub = rosNode.advertise('/chatter', std_msgs.String);
const msg = new std_msgs.String();
msg.data = 'Hello, 世界世界世界'
pub.publish(msg);
```
will fail with
```
[ERROR] [1587279960.400] (ros.rosnodejs): Error when publishing message on topic /chatter: RangeError [ERR_OUT_OF_RANGE]: The value of "length" is out of range. It must be >= 0 && <= 20. Received 22
    at Buffer.write (buffer.js:1037:7)
    at Object.StringSerializer [as string] (/workspace/ros/src/lib/node_modules/rosnodejs/dist/ros_msg_utils/lib/base_serialize.js:38:32)
    at Function.serialize (/opt/ros/kinetic/share/gennodejs/ros/std_msgs/msg/String.js:36:32)
    at Object.serializeMessage (/workspace/ros/src/lib/node_modules/rosnodejs/dist/utils/tcpros_utils.js:195:18)
    at /workspace/ros/src/lib/node_modules/rosnodejs/dist/lib/impl/PublisherImpl.js:224:43
    at Array.forEach (<anonymous>)
    at PublisherImpl._handleMsgQueue (/workspace/ros/src/lib/node_modules/rosnodejs/dist/lib/impl/PublisherImpl.js:219:16)
    at ClientQueue.handleClientMessages (/workspace/ros/src/lib/node_modules/rosnodejs/dist/utils/spinners/ClientQueue.js:46:20)
    at GlobalSpinner._handleQueue (/workspace/ros/src/lib/node_modules/rosnodejs/dist/utils/spinners/GlobalSpinner.js:174:24)
    at listOnTimeout (internal/timers.js:549:17)
RangeError [ERR_OUT_OF_RANGE]: The value of "length" is out of range. It must be >= 0 && <= 20. Received 22
    at Buffer.write (buffer.js:1037:7)
    at Object.StringSerializer [as string] (/workspace/ros/src/lib/node_modules/rosnodejs/dist/ros_msg_utils/lib/base_serialize.js:38:32)
    at Function.serialize (/opt/ros/kinetic/share/gennodejs/ros/std_msgs/msg/String.js:36:32)
    at Object.serializeMessage (/workspace/ros/src/lib/node_modules/rosnodejs/dist/utils/tcpros_utils.js:195:18)
    at /workspace/ros/src/lib/node_modules/rosnodejs/dist/lib/impl/PublisherImpl.js:224:43
    at Array.forEach (<anonymous>)
    at PublisherImpl._handleMsgQueue (/workspace/ros/src/lib/node_modules/rosnodejs/dist/lib/impl/PublisherImpl.js:219:16)
    at ClientQueue.handleClientMessages (/workspace/ros/src/lib/node_modules/rosnodejs/dist/utils/spinners/ClientQueue.js:46:20)
    at GlobalSpinner._handleQueue (/workspace/ros/src/lib/node_modules/rosnodejs/dist/utils/spinners/GlobalSpinner.js:174:24)
    at listOnTimeout (internal/timers.js:549:17)
```

I use ROS kinetic and rosnodejs 3.0.2. I cant figure out what gennodejs version is.

chris-smith, 2020-04-21:
I believe this is an issue with the version of gennodejs released with kinetic. It should be fixed if you overlay gennodejs in your catkin_workspace or if you use one of the other message generation options from rosnodejs.

chris-smith, 2020-06-05:
closing as this appears to be an issue with `gennodejs` which has a ticket to release

closed 2020-06-05 by chris-smith