RethinkRobotics-opensource/rosnodejs #151: remove master/slave references

trusktr, 2020-06-18:
It will be more respectful for all possible users to remove the `master` / `slave` terminology from the lib.

chfritz, 2020-06-18:
Would it make sense to wait and see how ROS core decides to resolve this issue first? I think it would be very confusing to use different terminology in the ROS libraries than in [ROS core](http://wiki.ros.org/ROS/Master_Slave_APIs).

trusktr, 2020-06-18:
I have a feeling it will take more time for ROS core, because it may be a huge breaking change. Do you know if there's interest there?

chfritz, 2020-06-18:
No, I haven't spoken with anyone at OSRF lately. I agree though that is could end up being a huge change. It might even require every ROS user to update their code, since the words in question appear in the API itself.

chris-smith, 2020-06-19:
I think I'd at least like to hear what ROS thinks. Doing this would break the api and make it harder for new and current users.