RethinkRobotics-opensource/rosnodejs #157: Messages don't actually come or go?

JamesNewton, 2020-08-11:
Warning: ROS Newbie. Also, I /think/ I'm an experienced Javascript programmer, which probably means I make stupid assumption based mistakes.

Issue: For some reason, messages don't pub or show up on sub despite the node showing up and the advertised topics appearing. 

I'm attempting to add ROS support to a javascript based robot w/o installing ROS on the robot as per my prior question at #131 

As documented at:
https://github.com/HaddingtonDynamics/Dexter/issues/97#issuecomment-667697296
so far I've been able to npm install the package, set the env vars, copy over the entire ROS share folder from my PC (TODO: Pair that down), and write a little test script that connects to my PC, subscribes to /chatter and advertises /chatter. 
````
const rosnodejs = require('rosnodejs');
//rosnodejs.loadAllPackages(); NO! Not for onTheFly / Kinetic and up.

rosnodejs.initNode('Dexter', { onTheFly: true
  ,rosMasterUri: `http://192.168.0.134:11311` //TODO: replace with real ROS master URI
  }).then(() => {
    console.log("Initialized")
    const nh = rosnodejs.nh;
    const stdMsgs = rosnodejs.require('std_msgs');
    const StringMsg = stdMsgs.msg.String;
    const sub = nh.subscribe('/chatter', StringMsg, (msg) => { console.log("recvd:"+msg+".") });
    const pub = nh.advertise('/chatter', StringMsg);
    pub.on('connection', () => {
      console.log("connected");
      pub.publish({data: "hi2\n"});
      });
    pub.publish({data: "hi1\n"});
    //const SensorMsgs = rosnodejs.require('sensor_msgs');
    //const joints = SensorMsgs.JointState;
    }
  );
````

And that appears to run correctly on the robot. e.g.
````
# node rostest.js
[INFO] [1597183568.704] (ros): Connected to master at http://192.168.0.134:11311!
Initialized
````
But notice that "connected" never fires. And neither "hi1" or "hi2" shows up on the PC even though I have `rostopic echo /chatter` running.

The node, and the topics, appear on my PC, including the ones from the robot. And my PC can see a publish on chatter from the PC. e.g.
````
$ roscore &
$ rostopic echo /chatter &
$ rosnode list
/Dexter
/rosout
/rostopic_11517_1597180072216
$ rostopic list
/chatter
/rosout
/rosout_agg
$ rostopic pub -1 /chatter std_msgs/String  -- "hello3"
publishing and latching message for 3.0 seconds
data: "hello3"
---
````

But on the robot side, the "hello3" message does NOT log to the console. 

I don't have any firewalls between myself and the bot, and even if I did, why would the node and topics show up and not the messages? 

Any ideas how to troubleshoot this?

chfritz, 2020-08-11:
You have one minor issue in your code: the type argument (second argument) to `advertise` and `subscribe` is a string not the object. Try:
```lang-js
    const sub = nh.subscribe('/chatter', 'std_msgs/String', (msg) => { console.log("recvd:"+msg+".") });
    const pub = nh.advertise('/chatter', 'std_msgs/String');
```

But my hunch is that this may not be the only problem. It may also (in addition) be an issue with your environment variables to support your multi-machine setup. It is actually *surprisingly* tricky to get them set up right, and it often -- at least for me -- involved editing your `/etc/hosts` file as well, because ROS is really picky about the way machines are named. I would recommend the following steps:
1. try running the rosnodejs code on your PC, i.e., with a local roscore, and make sure that works. It should after making the above changes.
2. try getting `rostopic echo` to run from the non-master. This is the real test for the environment setup. `rostopic list` may work fine, but `echo` won't, because `rostopic list` is handled by the roscore (XML protocol), while `echo` requires the rostcp/rosudp connection setup, which fails when the environment is not configured right. The pitfall is usually about how the master calls itself in ROSTCP messages and whether the client can resolve that name.
3. once that is working try again what you are trying now (again with the above changes to the code).

JamesNewton, 2020-08-11:
VERY nice answer! Thank you for being so quick and, as it turns out, I believe you are right about the host name thing. 
I did this on the PC
```
$ rosnode info /Dexter
--------------------------------------------------------------------------------
Node [/Dexter]
Publications: 
 * /chatter [std_msgs/String]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /chatter [std_msgs/String]

Services: 
 * /Dexter/get_loggers
 * /Dexter/set_logger_level

contacting node http://dex-197121:39327 ...
ERROR: Communication with node[http://dex-197121:39327] failed!

$ ping dex-197121
PING dex-197121 (92.242.140.2) 56(84) bytes of data.
^C
--- dex-197121 ping statistics ---
5 packets transmitted, 0 received, 100% packet loss, time 4091ms

nslookup dex-197121
Server:		127.0.1.1
Address:	127.0.1.1#53

Non-authoritative answer:
Name:	dex-197121
Address: 92.242.140.2
```
At which point my jaw fell off. Where the /heck/ would that IP address have come from? I cast about a bit and then just gave up and added a host entry with the robots actual IP. And as soon as I saved it, "connect" fired on the robot and "hi2\n" showed up on the PC. 

Now, when I publish a message to /chatter on the PC, I get:
````
[WARN] [1597185074.181] (ros.rosnodejs): Error requesting topic on /chatter: Error: getaddrinfo ENOTFOUND dex-197121 dex-197121:39327, undefined
````
And sure enough, the robot doesn't know the IP of it's host name... I'm sure if I tell it somehow, it will fix the issue. 

I'm also not an expert in Linux networking, so I have no idea what is going wrong there, but it's not a problem with rosnodejs. You wouldn't happen to be an Ubuntu 16.04 networking expert? ,o)

As to the use of an object or a string, that example was taken from your own documentation here:
https://github.com/RethinkRobotics-opensource/rosnodejs/blob/befb2ba5fdebb4f394a2a67813513c8a64a87217/README.md#L96

I've also made couple pull requests on the docs, hope that helps.

closed 2020-08-11 by JamesNewton

chfritz, 2020-08-11:
Regarding the use of a remote ROS master, I have put together this small script over the years so I can finally stop re-discovering (part of) the solution each time:
```sh
use_robot () 
{ 
    TARGET_IP=$1;
    IP=`ip route get $TARGET_IP | head -n 1 | sed "s/.*src \(\S*\) .*/\1/"`;
    echo -e "${GREEN}using $1 ($TARGET_IP) via $IP${NORMAL}";
    export ROS_MASTER_URI=http://$1:11311;
    export ROS_HOSTNAME=$IP;
    export ROS_IP=$IP
}
```
You use it on the non-master and call it with IP or hostname of the master as argument. That will take care of the ROS related environment variables that need to be set. This might be all you need at this point. If not then you may also need to add `dex-197121` to your `/etc/hosts` file on the robot on the line of localhost (there can be multiple names per line).

JamesNewton, 2020-08-11:
LOL. I'm /also/ not a Linux scripting expert. Is that a bash script? e.g. saved in a .sh file, chmod it to +x and then run it? I haven't seen the "use_robot () {" thing before... 

Just out of curiosity, is there any way you are aware of to automatically find the ROS "master" node when a robot connects to a new network? Does the server publish something that would allow other computers to find it?

chfritz, 2020-08-12:
That code snippet is a bash function definition. You can put it in your `.bashrc` and then, in a new shell, you can use that function. It's just a little more light-weight than a stand-alone bash script.

chfritz, 2020-08-12:
Regarding discover of the master: what would be neat! Something like zero-config. But no, I don't think that's currently possible. I think ROS across machines is still relatively uncommon and I guess you wouldn't want to rely on some auto-discovery in production anyhow, since it only opens up more ways to fail.

JamesNewton, 2020-08-12:
Well, this wasn't happening before, and now I can send messages from the robot to itself, and from the robot to the PC, but when I try to send messages from the PC to the robot, I get `[WARN] [1597209831.719] (ros.rosnodejs): Error requesting topic on /chatter: Error: getaddrinfo ENOTFOUND name_of_pc name_of_pc:38633, undefined`

Of course, the port number and time stamp are random numbers. 

And on the robot, I can not ping name_of_pc

But then, I don't understand how it even knows the name of the pc... It was subscribing to the pc via ip address... 

I've tried your script, and that allows me to not specify the rosMasterUri in the script, which is nice. But when I run the script, I'm seeing `using 192.168.1.134 (192.168.1.134) via 192.168.1.142` when I think I'm supposed to see name_of_pc in the () right? 
````
# ip route get 192.168.1.134
192.168.1.134 dev eth0  src 192.168.1.142 
    cache 
````
so that isn't returning the PC name.

I think the robot genuinely doesn't know the name of the pc, and then the message from the pc specifies the pc's name (?), and that causes the javascript to barf because it tries to return something and can't find the IP? 

Should I expand the script to take a second parameter of the PC's name (as well as it's IP) and set that into ROS_HOSTNAME? I can do that manually, but it actually makes things worse. The robot still can't ping the PC via its hostname.

I can put the PC's hostname in the robots /etc/hosts file, but it may change and that seems like a perminant solution to a temp problem.

chfritz, 2020-08-12:
No, this is one of the weird things in ROS. I was going down the same path you are thinking right now, but the problem is elsewhere. In ROS, each node reports to the master with the services and topics it provides. When doing that they use a name, and that's the name their topic will be offered under. The roscore will give subscriber that name. The name nodes use is usually their hostname but you can override it by setting ROS_HOSTNAME. So I think in your case you can either: add name_of_pc to `/etc/hosts` on the robot, or set ROS_HOSTNAME on the PC to something the robot can resolve.

JamesNewton, 2020-08-13:
Can ROS_HOSTNAME on the PC be set to the PC's IP address?

chfritz, 2020-08-13:
I think so, yes.

JamesNewton, 2020-08-13:
And yes, that does work... so if each side uses it's own IP as the ROS_HOSTNAME, then the other side won't have problems responding to them...
OH! That's what the `ip route` line in your script is doing! LOL. Duh. I totally didn't get that until just now. I thought it was trying to find the hostname of the 'master'. 

Now I just have to figure out why the exports from your script aren't being retained... it finds the right values, but then when I check after running it, the variables aren't set. 

BTW: Thank you again, SO MUCH, for all your help with this and for responding so quickly.