RethinkRobotics-opensource/rosnodejs #160: Cannot Implement Turtlesim Actionlib example

Kasidit0052, 2020-09-11:
Hello, I've encountered some problem regarding the turtle_actionlib shape_client examples. Even though the script detect roscore, the actionlib server receives nothing

This is the output log
`[INFO] [1599833315.613] (ros): Connected to master at http://localhost:11311!`

and this is my client-side code
```
#!/usr/bin/env node
'use strict';
let rosnodejs = require('rosnodejs');
rosnodejs.initNode('/my_node', {onTheFly: true}).then((rosNode) => {
  const shapeActionGoal = rosnodejs.require('turtle_actionlib').msg.ShapeActionGoal;
  const ac = rosnodejs.nh.actionClientInterface(
    "/turtle_shape",
    "turtle_actionlib/Shape"
  );
  ac.sendGoal(new shapeActionGoal({
    goal: {
        edges: 3,
        radius: 1
      }
    }));
  setTimeout(function() {
    console.log("timeout")
    ac.cancel();
  }, 1000);

});

```
If the problems can be solved, could you please elaborate the solution?
Thank you

chfritz, 2020-09-14:
Confirmed, we'll need to debug this. Note however than when run with some delay (1000ms or so) it seems to work, which is also what the example code in the repo does.

chris-smith, 2020-09-15:
I think the issue here is 1) there's no waiting for the action server to connect and 2) the published topics are not latched.

Kasidit0052, 2020-09-15:
Thank you for your response, I have already included the setTimeout function as you suggested.
```
  setTimeout(function() {
    const shapeActionGoal = rosnodejs.require('turtle_actionlib').msg.ShapeActionGoal;
    const ac = rosnodejs.nh.actionClientInterface(
      "/turtle_shape",
      "turtle_actionlib/Shape"
    );
    ac.sendGoal(new shapeActionGoal({
      goal: {
          edges: 3,
          radius: 1
        }
      }));
    setTimeout(function() {
      console.log("timeout")
      ac.cancel();
    }, 1000);
  }, 2000);
```
However, the problem still exists. So, "the published topics are not latched" might be the main reason. Unfortunately I don't know how to enable latching in rosnodejs actionClientInterface.  Do I need to add an additional argument?

chfritz, 2020-09-15:
As a fall-back, for now, you can also just publish the action topics (goal and cancel) and subscribe to feedback, status, and result directly. That will work, just like it does for any other topics.

chris-smith, 2020-09-15:
Latching can be passed when constructing the action client
```
rosnodejs.nh.actionClientInterface('/turtle_shape', 'turtle_actionlib/Shape', { goal: { latching: true } });
```
I think the waiting needs to occur between creating the interface and sending the goal. Either just sleep for a bit or you can call `ac. waitForActionServerToStart();`.

It looks to me like the example likely hasn't worked in a long time because it's not waiting and there's no latching. The action clients used to latch by default but it was changed to match the C++ implementation and we forgot to update the example.

chris-smith, 2020-09-28:
I'm going to assume this is resolved for now. I've opened #162 to address the examples not working. Feel free to reopen if necessary.

closed 2020-09-28 by chris-smith