RethinkRobotics-opensource/rosnodejs #173: xmlrpc lib crash ECONNRESET?

panshengjie, 2021-03-25:
`xmlrpc/lib/server.js `

handleMethodCall  and httpServer  not handle error, will this be fatal?

I've being tracking  random crash issue for quite a long time , lack of error handle seems to crash the node

chfritz, 2021-03-25:
hm, I've never come across this in several years of using rosnodejs in production. Can you say more about these random crash issues? Have you found a way to reproduce them?

panshengjie, 2021-03-25:
pretty hard to reproduce . our robot runs all day&night. It seems crash happens in a level : 100 robots crash 1 in  2 weeks

panshengjie, 2021-03-25:
in addition, maybe some of out program didn't use ROS correctly, sending to much msgs and services. rasie rosmaster to 10-15% CPU. and some client may crash and reset the RPC/connection. could be a race condition. Here is some refs I found so far. [https://github.com/baalexander/node-xmlrpc/pull/82](url)
[https://github.com/baalexander/node-xmlrpc/issues/143](url)
[https://github.com/nodejs/node/issues/27916](url)

chris-smith, 2021-04-16:
Haven't seen this either. We could add an error handler on our end, though it's not clear to me what state things would be in afterwards... I'm assuming the server is emitting an 'error' event when this happens?

panshengjie, 2021-04-19:
I've added error & clientError handle in httpServer (at xmlrpc server.js ), but has no effects. Is there any where should I add error handle?

panshengjie, 2021-05-31:
finnaly，maybe I've got this.
https://github.com/RethinkRobotics-opensource/rosnodejs/blob/8aefc938a34c40131dad99d85252f698d10f191e/src/lib/RosNode.js#L452

the connection instance not have error handle.
some time it has a error handle, since  I saw some other error log . some time it not, and may crash the whole program

closed 2021-06-01 by panshengjie