RethinkRobotics-opensource/rosnodejs #175: Is it possible to receive message without deserialize?

lantica, 2021-04-15:
I'd like to subscribe to a topic to serve as health checking.
However, the message being published is quite large and it leads to a high usage of CPU during deserialization of message which the content of the message doesn't matter to me.

Is there any way to receive the message without deserialization?

chris-smith, 2021-04-16:
You could hack your way around to it. [This](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/lib/impl/SubscriberImpl.js#L482) is where we attach the event listener that leads to deserialization. You could remove the listener we attach and add your own. There's one of these per publisher connection.

Though if you have access, I'd recommend just adding another topic you can subscribe to.

lantica, 2021-04-16:
Thank you for the advice. I can add my on event listener on it now.

However, the cpu usage does not reduce even it does not deserialize the message.
Could you suggest some other possibilities causing the high cpu usage?
The message type of the topic I subscribing to is PointCloud2.

chris-smith, 2021-04-17:
The only other thing I can think of is just from pulling the messages off the wire - that happens [here](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/utils/serialization_utils.js#L58). PointCloud messages are generally pretty big and we don't reuse buffers so it could cause a lot of memory allocation.

chfritz, 2021-04-17:
That issue seem to be recurring, and reusing buffers was also suggested here: https://github.com/RethinkRobotics-opensource/rosnodejs/issues/118#issuecomment-639252354. Seems like a good idea.

closed 2021-05-14 by lantica