RethinkRobotics-opensource/rosnodejs #178: How to registry a new msg. How to subscribe custom messages? I need help!

f-y-l, 2021-06-21:
my.js
```
const gui_msgs = rosnodejs.require('gui_msgs').msg;
let sub2 = rosNode.subscribe('/mode_cmd', gui_msgs.mode_cmd,
    (data) => { // define callback execution
        rosnodejs.log.info('I heard /mode_cmd: [' + data + ']');
    }
);
```
rosnodejs>dist>utils>messageGeneration>messages.js
```
// Registry

var registry = {
  'gui_msgs':{
    msg:{
      'mode_cmd':{
        'header':"Header",
        'mode':"int32"
      }
    },
    srv:{
      Request:{},
      Response:{}
    }
  }
};
/*
   registry looks like:
  { 'packagename':
    {
      msg: {
        'String': classdef,
        'Pose': classdef,
        ...
      },
      srv: { Request:
             {
               'SetBool': classdef,
               ...
             },
             Response:
             {
               'SetBool': classdef,
               ...
             }
           }
    },
    'packagename2': {..}
  };
*/
```
rosbagmsg:
[image]
error:
[image]

chfritz, 2021-06-21:
which message generation are you using? [`onTheFly`](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/examples/on_the_fly.js#L5) or the default, provided by gennodejs?

f-y-l, 2021-06-22:
Thank you for your answer. I created a new MSG in my workspace. After compiling, it can be solved by running

closed 2021-06-22 by f-y-l

chengchenglee, 2022-04-14:
@f-y-l 
I am facing the same problem could you please provide more details about the solution?