RethinkRobotics-opensource/rosnodejs #187: Unknown client Subscriber error, from GlobalSpinner

SirFailaL0t, 2022-03-30:
Rosnodejs was running fine for a while but I started getting this error:

`Error: Unable to queue message for unknown client Subscriber:///points_concat`
  `at GlobalSpinner._queueMessage (.../rosnodejs/dist/utils/spinners/GlobalSpinner.js:118:13)`
  `at GlobalSpinner.ping (.../rosnodejs/dist/utils/spinners/GlobalSpinner.js:98:12)`
  `at SubscriberImpl._handleMessage (.../rosnodejs/dist/lib/impl/SubscriberImpl.js:402:37)`

despite not having changed much from the working state.
The nodes do subscribe to the configured topics, but I can't use throttleMs or queueSizes other than -1.
Could this have something to do with me switching from Node version 10 to 12?
Switching back to Node version 10 is not as easy as I thought, in my project, so I can't test that right now.

chfritz, 2022-03-30:
That client name looks strange. What are you publishing and what are you using to subscribe to it? Would you be able to provide a minimal example to reproduce this? It would greatly help in debugging this.

chris-smith, 2022-03-30:
I'd be surprised if this is related to node version... the `Subscriber://` thing is something we prefix to topic names when adding items to the spinner.

SirFailaL0t, 2022-03-30:
The /points_concat topic consists of sensor_msgs/PointCloud2 messages.
`import rosnodejs from 'rosnodejs';`
`rosnodejs.initNode('listener', {onTheFly: false}).then(rosNode => {`
      `for (const tc of this.rosConfig.topicConfig) {`
          `this.listeners[tc.topic] = rosNode.subscribe(`
           `tc.topic, tc.type, msg => this._onMessage(tc.topic, msg),`
            `{queueSize: 0, throttleMs: 0}) } } )`
This is how I subscribed. The tc.topic and tc.type for the error above were "/points_concat" and "sensor_msgs/PointCloud2".
But the error triggers for the first message I guess because it fails on different topics and message types,
not only "/points_concat". It also failed for sensor_msgs/Image, PointStamped and every other type.
Using `{queueSize: -1, throttleMs: -1}` works though.
The data is coming from a rosbag file that I'm playing.

SirFailaL0t, 2022-03-30:
I was able to fix and reproduce this error. As far as I found out when throttleMs is 0, queueSizes -1 and 0 can't be used 
or this error comes up. Took me a while to realize that this small change was responsible :"D Didn't think much about what values to use beforehand. Can one of you reproduce the error like that?

chris-smith, 2022-03-30:
I'd guess this also happens for any `throttleMs` value `>= 0`? `throttleMs` less than 0 is used to indicate that message callbacks should be invoked immediately without any delay. `queueSize` of 0 should maybe indicate the same thing, but it doesn't work that way right now. Looks like `queueSize < 1` was partially intended to [indicate an infinite queue ](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/utils/spinners/ClientQueue.js#L11)but we also explicitly [check](https://github.com/RethinkRobotics-opensource/rosnodejs/blob/kinetic-devel/src/utils/spinners/GlobalSpinner.js#L71) that the queue size is greater than 0 so it will never work that way.

Seems like something to fix though I'm not sure it will change the range of available behaviors, just provide errors or reasonable behavior for cases that currently trigger this.

SirFailaL0t, 2022-03-31:
Tested it again with `throttleMs=-1` and all `queueSize` values worked. For me 0 for no queue at all and -1 for
infinite queue(if this is a use case), would feel intuitive.

closed 2022-03-31 by SirFailaL0t