RethinkRobotics-opensource/rosnodejs #191: onTheFly error

chengchenglee, 2022-04-14:
I run my program successfully.

 adversely, when I run onTheFly, some errors appear.
could any help?

function talker() {
  rosnodejs.initNode('my_node', { onTheFly: true })
  .then((rosNode) => {
MY PROGRAME
}

Thanks in advance!

Error while parsing message loc_svcs_mm/pose: Error: Invalid empty type undefined
    at parseType (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:57:11)
    at MsgSpec._parseLine (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:439:28)
    at Array.forEach (<anonymous>)
    at MsgSpec._parseMessage (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:379:15)
    at MsgSpec.loadFile (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:190:14)
    at new MsgSpec (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:360:11)
    at Function.create (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:326:18)
    at /home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:363:33
    at /home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:352:23
    at Array.forEach (<anonymous>)
Error: Invalid empty type undefined
    at parseType (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:57:11)
    at MsgSpec._parseLine (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:439:28)
    at Array.forEach (<anonymous>)
    at MsgSpec._parseMessage (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:379:15)
    at MsgSpec.loadFile (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:190:14)
    at new MsgSpec (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:360:11)
    at Function.create (/home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:326:18)
    at /home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:363:33
    at /home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:352:23
    at Array.forEach (<anonymous>)
[ERROR] [1649916950.060] (ros): Error during initialization: Error: Invalid empty type undefined
(node:25947) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'advertise' of undefined
    at /home/abcde/workspace/rosnodejs/src/rosnodejs/js_module/scripts/ros_node_script.js:97:30
(Use `node --trace-warnings ...` to show where the warning was created)
(node:25947) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:25947) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

chfritz, 2022-04-14:
The error you show is not caused by the code you shared. Can you please post the code where you are using `loc_svcs_mm/pose` and also the result of `find /opt/ros/*/share/loc_svcs_mm` and `echo $CMAKE_PREFIX_PATH`?

chengchenglee, 2022-04-15:
Hello @chfritz 

I would like to provide more information here.

my code is below.

============================
#!/usr/bin/env node

'use strict';
const rosnodejs = require('rosnodejs');
const custom_messages_module = rosnodejs.require('custom_messages_module').msg;
const std_msgs  = rosnodejs.require('std_msgs').msg;

console.log("- Loaded expected nodes");

var test =new custom_messages_module.gpsState();

function talker() {
rosnodejs.initNode('my_node', { onTheFly: true })
  .then((rosNode) => {
    const custom_messages_module = rosnodejs.require('custom_messages_module').msg;
    
  });
}
if (require.main === module) {
  // Invoke Main Talker Function
  talker();
}

============================

you can see that the code is quite simple(almost same as the sample code)
then I get the error log as below

============================
Error while parsing message loc_svcs_mm/pose: Error: Invalid empty type undefined
    at parseType (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:57:11)
    at MsgSpec._parseLine (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:439:28)
    at Array.forEach (<anonymous>)
    at MsgSpec._parseMessage (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:379:15)
    at MsgSpec.loadFile (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:190:14)
    at new MsgSpec (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:360:11)
    at Function.create (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:326:18)
    at /home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:363:33
    at /home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:352:23
    at Array.forEach (<anonymous>)
Error: Invalid empty type undefined
    at parseType (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:57:11)
    at MsgSpec._parseLine (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:439:28)
    at Array.forEach (<anonymous>)
    at MsgSpec._parseMessage (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:379:15)
    at MsgSpec.loadFile (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:190:14)
    at new MsgSpec (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:360:11)
    at Function.create (/home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageSpec.js:326:18)
    at /home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:363:33
    at /home/abcde/workspace/rosnodejs/src/js_module/node_modules/rosnodejs/dist/utils/messageGeneration/MessageLoader.js:352:23
    at Array.forEach (<anonymous>)
[ERROR] [1650034672.296] (ros): Error during initialization: Error: Invalid empty type undefined

============================

A strange thing is that I even could not succesfully execute the find command.
under share folder, I type loc and press tab key several times, no candidate appears.
does it mean I miss some library in my PC?

============================
~$ find /opt/ros/kinetic/share/loc_svcs_mm
find: ‘/opt/ros/kinetic/share/loc_svcs_mm’: No such file or directory

============================

finally my env parameter is here.

============================
~$ echo $CMAKE_PREFIX_PATH
/home/abcde/workspace/rosnodejs/devel:/home/abcde/projects/nrc_ws/devel:/home/abcde/projects/externals_ws/devel:/opt/ros/kinetic

============================

If you need more details for investigation, please feel free to leave comment
I will provide all information you need.

Thank you a lot, sincerely!

chfritz, 2022-04-15:
I'm still confused. Your code doesn't mention the custom message type `loc_svcs_mm/pose` at all. Could you show the content of the `custom_messages_module` package? (presumably in `/opt/ros/kinetic/share/custom_messages_module`).

BTW, it would help a lot if you could format your code as such. Just add "```js" before a code block and end it with "```".

chengchenglee, 2022-04-15:
Hello @chfritz again

I push my code to my git for your reference.

https://github.com/chengchenglee/sharingcode

my env is ubuntu 16.04 ros kinetic

if any more info is needed, please leave comments.

Thanks as always!

chengchenglee, 2022-04-16:
Hello @chfritz 
I check my environment.
there is even no such a directory
/opt/ros/kinetic/share/custom_messages_module

I re-post my code here.
Apologize that I am not quite familiar with GitHub.

```js
#!/usr/bin/env node

'use strict';
const rosnodejs = require('rosnodejs');
const custom_messages_module = rosnodejs.require('custom_messages_module').msg;
const std_msgs = rosnodejs.require('std_msgs').msg;

console.log("- Loaded expected nodes");

var test =new custom_messages_module.gpsState();

function talker() {
rosnodejs.initNode('my_node', { onTheFly: true })
.then((rosNode) => {
const custom_messages_module = rosnodejs.require('custom_messages_module').msg;

});
}
if (require.main === module) {
// Invoke Main Talker Function
talker();
}
```

chengchenglee, 2022-04-19:
> I'm still confused. Your code doesn't mention the custom message type `loc_svcs_mm/pose` at all. Could you show the content of the `custom_messages_module` package? (presumably in [...]

@chfritz
I do not know in detail, how onthefly works.

`/opt/ros/kinetic/share/` I think under this directory there are some packages  installed in the OS.

but the `custom_messages_module` is the folder that under ros `src` folder which created by myself.
```js
const custom_messages_module = rosnodejs.require('custom_messages_module').msg;
```
the command above could also access the module 
`devel/share/gennodejs/ros`

if possible, could you please share a good example by using onTheFly?
also, is this problem related to gennodejs?

chengchenglee, 2022-05-10:
onthefly is not a must function and close this topic

closed 2022-05-10 by chengchenglee