RethinkRobotics-opensource/rosnodejs #192: _index.js items different from cmake add_message_files items

chengchenglee, 2022-04-14:
Hello, I try to use customized message type to transfer messages,
I happened to meet some error when using rosnodejs.require to call some packages.

I trace the reason and find out there is a strange thing as shown below.

first I add msg files to CMakeLists,txt

add_message_files(FILES
  gpsState.msg
  TrackedObject.msg
  TrackedObjectSet.msg
)

then I catkin_make my project and check the file in devel folder.

workspace/devel/share/gennodejs/my_project_name/msg/_index.js

"use strict";

let TrackedObject = require('./TrackedObject.js');
let TrackedObjectSet = require('./TrackedObjectSet.js');
let gpsState = require('./gpsState.js');
_let DynamicPoseWithCovar = require('./DynamicPoseWithCovar.js');_

module.exports = {
  TrackedObject: TrackedObject,
  TrackedObjectSet: TrackedObjectSet,
  gpsState: gpsState,
  _DynamicPoseWithCovar: DynamicPoseWithCovar,_
};

the words in italic is not defined in my CMakeLists file.
and it leads to the failure when I want to use other type messages.

Could anyone could help?

Thanks in advance!

chfritz, 2022-04-14:
don't you just need to add `DynamicPoseWithCovar.msg` to your `CMakeLists.txt`?

chengchenglee, 2022-04-15:
Hello @chfritz 

I think I know the reason now.

yes, you are right.
if I just add **DynamicPoseWithCovar.msg** to my **CMakeLists.txt**, everything is fine.

but here there is a very tricky thing I found.
in the dir **workspace/devel/share/gennodejs/my_project_name/msg/**

_index.js will list all messages files which exist in src/ msg folder,
however, other *.js files are generated based on your description in **CMakeLists.txt**  **add_message_files**.

**add_message_files** has to be identical with really files in **msg** folder.

it means even though you put some message files in msg folder, not quoted in **CMakeLists.txt,**
then when you want to require that package, error will pop up.

I know how to solve this problem.
well, I will not say it is a bug, but it is a little tricky,

 Thanks a million!

closed 2022-04-15 by chengchenglee