RethinkRobotics-opensource/rosnodejs #193: TypeError: Cannot read property 'msg' of undefined

chengchenglee, 2022-04-18:
I happen to an error like this.

rosrun nrc_svcs http_ros.js 
/home/abcde/projects/nrc_ws/src/nrc_svcs/src/infra_http/scripts/http_ros.js:11
const TOS = rosnodejs.require('nrc_msgs').msg;
                                         ^

TypeError: Cannot read property 'msg' of undefined
    at Object.<anonymous> (/home/abcde/projects/nrc_ws/src/nrc_svcs/src/infra_http/scripts/http_ros.js:11:42)
    at Module._compile (internal/modules/cjs/loader.js:1085:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
    at Module.load (internal/modules/cjs/loader.js:950:32)
    at Function.Module._load (internal/modules/cjs/loader.js:790:12)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12)
    at internal/main/run_main_module.js:17:47

/dev/share/gennodejs/ros/nrc_msgs/srv/_index.js

```js
"use strict";

module.exports = {
  msg: require('./msg/_index.js'),
  srv: require('./srv/_index.js')
};
```

and in my srv folder is as below.

[image]

if I delete the line
```js
  srv: require('./srv/_index.js')
```

everything is fine.

Could someone help me?

Thanks

chfritz, 2022-04-18:
do you have any `.srv` files? If yes, can you post them?

chengchenglee, 2022-04-19:
@chfritz Thank you as always!

I check my `.srv` files one by one, and I find out that there is a dependency between `.srv` file  and other `.msg` files.

unfortunately, I manually delete those `.msg` files, so that when program runs the line.
in GetMapAt.js file
```js
let MapLaneSet = require('../msg/MapLaneSet.js');
```
the program could not find out  `../msg/MapLaneSet.js`<=this file.
that is where the error comes from.

A good lesson for me. :)

closed 2022-04-19 by chengchenglee