RethinkRobotics-opensource/rosnodejs #197: Deal With Custom ROS Messages

CrawlerWheel, 2022-10-18:
Hi，author！How can I parse ros messages defined by myself in nodejs with your package？
For example，when I wanna subscribe some topic in nodejs to get my custom  ros message type Target:
`int32 id float64 prob int32 x int32 y` and type TargetArray:`Header header Target[] targets`.
By the wey ,these custom ros messages are used to describe some yolo detection results.

chfritz, 2022-10-18:
So long as the `.msg` files describing your custom messages are in a package inside a folder included in your `ROS_PACKAGE_PATH` env var you can use the `onTheFly` option to load it. That will parse and load them at runtime. Alternatively you can include your custom package in your catkin workspace so that `gennodejs` picks them up and compiles them at `catkin_make` time.

CrawlerWheel, 2022-10-19:
Thanks very much!

closed 2022-10-19 by CrawlerWheel